FreeCAD/FreeCAD #16109: Draw Style: No option to view hidden lines (Flat lines with hidden edges)

maxwxyz, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently there is no shading option in the draw styles menu to display the objects in the 3D view including the hidden edges as it is used in TechDraw for example.
There is a draw style "hidden lines" but it displays the triangulation of the model. The desired result would be a mix of "Wireframe" and "Shaded".

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

northcoastam, 2024-09-17:
Can confirm the same behaviour on RC1 
[image]

OS: 
Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
Installed mods: 
  * A2plus 0.4.65
  * A2plus.backup1693449725.06262 (Disabled)
  * Curves 0.6.42
  * Curves.backup1693449730.5624967 0.6.13 (Disabled)
  * Defeaturing 1.2.2
  * Defeaturing.backup1693449735.7019093 (Disabled)
  * fasteners.backup1695592281.051254 0.4.63 (Disabled)
  * Freecad-Built-in-themes-beta 1.2.2
  * FreeCAD-themes 2024.7.24
  * lattice2 1.0.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
```

```

maxwxyz, 2025-06-05:
@kadet1090 maybe interesting for you for selection order

luzpaz, 2025-06-15:
Upstream Coin has merged the PR

tetektoza, 2025-06-16:
@luzpaz I've seen. Maybe we can wait a little bit, as I'm doing this one: https://github.com/FreeCAD/FreeCAD/issues/10614
And I think it will also require changes in coin but I need to see more. Basically, we could have both features in one bumped version, so let's hold on. I have implementation ready for hidden shaded lines on our side already, so no problem.