FreeCAD/FreeCAD #16113: OriginGroupExtention::hasObject does not work for link groups.

PaddleStroke, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

- part
- - linkGroup
- - - elt1

`part->hasObject(elt1, true)`

returns false when it should return true.

@wwmayer 

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-08-28:
I am not sure what is the best way to fix it. Should we add a else if to GroupExtension::hasObject to test if child is linkgroup and if so check the elements?

wwmayer, 2024-08-28:
I think the behaviour is correct but your expectation is wrong. If an item in the tree view is displayed as a child or grand child item -- which is controlled by the view provider --  does not necessarily imply that the related object belongs to the group object.

To see what I mean it's best to display the dependency graph. Here you will see that the object (e.g. elt1) is outside the part container.

For a recursive search the method `GroupExtension::hasObject` simply checks if the passed object also has a GroupExtension but this is not the case for a LinkGroup which has a LinkBaseExtension.

PaddleStroke, 2025-07-17:
still relevant

FEA-eng, 2026-11-02:
What's `elt1`? Any object within LinkGroup named this way? I get:

```
>>> obj.hasObject(elt1, true)
Traceback (most recent call last):
  File "<input>", line 1, in <module>
NameError: name 'elt1' is not defined
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```