FreeCAD/FreeCAD #16116: Crash when dimline location is 0?

outpaddling, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Running the latest commit from Main (73cb5bafe0), I created a new dimension for an edge using draft workbench.  I can then go to the dimline properties and change the x and z locations, which are not zero.  However, when I try to change y, which is 0, FreeCAD crashes.

Below is the standard output after selecting the y location:

```
ASSERT: "!(P(max) < P(min))" in file /usr/local/include/qt6/QtCore/qminmax.h, line 75                                                                           
Abort (core dumped)                                                             
```

Standard output after creating the dimension (I ran FreeCAD from a terminal):

```
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex4         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex3         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex2         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex5         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex7         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex5         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex3         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex9         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex4         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex8         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex10        
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex10        
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge1           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge1           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge5           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge11          
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge4           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge13          
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge8           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex4         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex3         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex4         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex6         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex7         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex2         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex3         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex7         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex4         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex10        
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex11        
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex9         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex12        
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex10        
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex12        
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge1           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge7           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge2           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge3           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge4           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge5           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge14          
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Edge7           
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex1         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex4         
<TopoShape> TopoShapeExpansion.cpp(1919): unnamed lower element Vertex3         
[... 309 lines cut ...]
    frame #46: 0x0000000826a8594e libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2159:5
    frame #47: 0x000000000022cfda FreeCAD`main(argc=1, argv=0x0000000820e766c8) at MainGui.cpp:300:13
    frame #48: 0x000000083d1b2afa libc.so.7`__libc_start1 + 298
    frame #49: 0x000000000022ad90 FreeCAD`_start at crt1_s.S:83
```

### Full version info

```shell
I learned from another issue that FreeCAD's "About" feature reports the wrong revision number, so providing git commit info instead:

# git branch
* main

# git describe --tags
0.21rc1-4887-g73cb5bafe0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

outpaddling, 2024-08-28:
> @outpaddling is this the same behavior in the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds?

Not sure what you're asking.  I'm running the latest commit from Main branch.

> Please provide full about info and a demo file.

Well, like I said, the version info in "About" is wrong (see discussion in 16047), but here it is anyway:

OS: FreeBSD 14.0-RELEASE-p9 (Lumina/Lumina)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.14555 (Git shallow)
Build type: Release
Branch: master
Hash: 4fa2ed7a320e49cf803c7f6a81db9084220bf453
Python 3.11.9, Qt 6.7.2, Coin 4.0.2, Vtk 9.2.6, OCC 7.8.1
Locale: C/Default (C)
Installed mods: 
  * ArchTextures

maxwxyz, 2024-08-28:
ok, did it work in the stable 0.21.2 version?

outpaddling, 2024-08-28:
0.21.2 has never crashed for me.   I'm not a sophisticated CAD user, but I've been using it regularly for several months and it seems to work as expected.

maxwxyz, 2024-08-28:
@Roy-043 FYI

outpaddling, 2024-08-28:
Attaching a model to help reproduce the issue.

[open-porch.zip](https://github.com/user-attachments/files/16789210/open-porch.zip)

1. Go to Front view
2. Go to Draft toolbox
3. Select top edge of part "Gate"
4. Create a dimension
5. Go to Dimline properties (My case: x = 2'3"+7/8", y = 0, z = 11'2" + 5/8")
6. Click on y -> crash.  Clicking on x or z works fine.

Side note: Github is not accepting a TGZ file, even though it says it does.  I had to use zip instead.

```
# file 
open-porch.tgz 
open-porch.tgz: gzip compressed data, last modified: Wed Aug 28 21:36:15 2024, from Unix, original size modulo 2^32 200192
# tar ztf open-porch.tgz 
open-porch.FCStd
```

outpaddling, 2024-08-28:
The problem is not limited to dimension lines.
Same thing happens if I change the the x, y, or z value of a part position to 0.

```
Power-line: Rotation angle of cylinder too small                                
Power-line: Rotation angle of cylinder too small                                
ASSERT: "!(P(max) < P(min))" in file /usr/local/include/qt6/QtCore/qminmax.h, line 75                                                                           
Abort (core dumped)                                                             
```

Roy-043, 2024-08-29:
According to @luzpaz we support Qt6:
https://github.com/FreeCAD/FreeCAD/issues/16047#issuecomment-2309140014

I am not sure this is in fact the case. Was there an official announcement?

outpaddling, 2024-08-29:
FYI, I corrected the title of the issue again.  The crash does not occur when changing *to* 0, but when trying to change *from* 0.

luzpaz, 2024-08-29:
@Roy-043 hmm..i was under the impression we are building qith Qt6 as well, no ?

mr-miky, 2024-08-30:
> FWIW, this issue and #16047 (which also happens with Qt5) are the only problems I've seen so far with the Qt6 build. I haven't check the Qt5 build for this issue, since I only stumbled on it after switching to Qt6.

> Attaching a model to help reproduce the issue.
>
> [open-porch.zip](https://github.com/user-attachments/files/16789210/open-porch.zip)
>
>     1. Go to Front view
>
>     2. Go to Draft toolbox
>
>     3. [...]

I tried to do what you suggest with the dim line but nothing unusual happens.
I tried to move some parts to 0,0,0 and then bring them back in the original position but even in this case nothing strange happens.

OS: Fedora Linux 40 (MATE-Compiz) (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.pre_38555.fc40 (Git)
Build type: Release
Python 3.12.5, Qt 6.7.2, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Italian/Italy (it_IT) ]

xtemp09, 2024-08-30:
This issue and #16047 have one thing in common ― the OS and the fact that the crashes are not reproducible on other OSes.

@outpaddling, could you, at least, provide some information about the compiler and the configuration options? _FreeCAD_ cannot be compiled right away with _clang_, it requires some options to be set. I set `-Wno-enum-constexpr-conversion` to compile on _KUbuntu 24_:

```
Ubuntu clang version 18.1.8 (++20240720065638+3b5b5c1ec4a3-1~exp1~20240720065751.142)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin 
```

<details><summary>conf</summary>

```
cmake ../freecad-source -G Ninja \
-D CMAKE_BUILD_TYPE=Debug \
-D CMAKE_INSTALL_PREFIX=/home/xtemp09/fc-test/ \
\
-D CMAKE_C_COMPILER=clang -D CMAKE_CXX_COMPILER=clang++     \
-D CMAKE_C_FLAGS_DEBUG="-glldb -Og -fno-omit-frame-pointer -Wno-enum-constexpr-conversion"   \
-D CMAKE_CXX_FLAGS_DEBUG="-glldb -Og -fno-omit-frame-pointer -Wno-enum-constexpr-conversion" \
\
-D CMAKE_EXE_LINKER_FLAGS="-fuse-ld=lld"    \
-D CMAKE_SHARED_LINKER_FLAGS="-fuse-ld=lld" \
\
-D CMAKE_AR=/usr/bin/llvm-ar-18               \
-D CMAKE_ADDR2LINE=/usr/bin/llvm-addr2line-18 \
-D CMAKE_MT=/usr/bin/llvm-mt-18               \
-D CMAKE_NM=/usr/bin/llvm-nm-18               \
-D CMAKE_OBJCOPY=/usr/bin/llvm-objcopy-18     \
-D CMAKE_OBJDUMP=/usr/bin/llvm-objdump-18     \
-D CMAKE_RANLIB=/usr/bin/llvm-ranlib-18       \
-D CMAKE_READELF=/usr/bin/llvm-readelf-18     \
-D CMAKE_RC_COMPILER=/usr/bin/llvm-rc-18      \
-D CMAKE_STRIP=/usr/bin/llvm-strip-18
```

</details>

outpaddling, 2024-08-31:
Yeah, could be so many things.  My first guess would be a simple bug that manifests differently on different platforms due to stack/heap organization, malloc() behavior, etc.  I run into this all the time testing my own code on various platforms.  E.g., a user of one of my programs recently reported an crash that only happened on recent Linux systems.  I had run it extensively on Linux a few years back and could not reproduce the problem on BSD or Mac.  Turned out to be an uninitialized structure member that happened to always get 0 on BSD and Mac, and most of the time on Linux.

Next guess would be the specific version or build options of one of the deps.

I tried adding -Wno-enum-constexpr-conversion, but it had no effect.  Note that 0.21.2 is built using clang without -Wno-enum-constexpr-conversion, and it works fine.  Must be something specific to clang + Kubuntu.

I'll poke around in the code myself.  I don't expect to understand it very well, but I might get lucky and spot something suspicious.

Inserting my full config output below.

```
===>  Configuring for FreeCAD-0.21.2.4887
===>  Performing out-of-source build
/bin/mkdir -p /usr/ports/wip/freecad/work/.build
-- The C compiler identification is Clang 16.0.6
-- The CXX compiler identification is Clang 16.0.6
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiler: Clang, version: 16.0.6
-- Looking for GL/gl.h
-- Looking for GL/gl.h - found
-- Looking for C++ include istream
-- Looking for C++ include istream - found
-- Looking for C++ include ostream
-- Looking for C++ include ostream - found
-- Looking for C++ include fstream
-- Looking for C++ include fstream - found
-- Looking for C++ include sstream
-- Looking for C++ include sstream - found
-- Looking for C++ include ios
-- Looking for C++ include ios - found
-- Looking for C++ include iostream
-- Looking for C++ include iostream - found
-- Looking for C++ include iomanip
-- Looking for C++ include iomanip - found
-- Looking for C++ include iostream
-- Looking for C++ include iostream - found
-- Check for STD namespace
-- Check for STD namespace - found
-- Force BOOST_PP_VARIADICS=1 for clang
-- prefix: /usr/local/FreeCAD
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.30.2
-- Compiling with Qt 6
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.9.6") found components: doxygen dot
-- find_package() was used to locate fmt version 10.2.1
-- Found Python3: /usr/local/bin/python3.11 (found version "3.11.9") found components: Interpreter Development Development.Module Development.Embed
-- Performing Test HAS_FLTO_THIN
-- Performing Test HAS_FLTO_THIN - Success
-- Found pybind11: /usr/local/lib/python3.11/site-packages/pybind11/include (found version "2.13.5")
-- Found XercesC: /usr/local/lib/libxerces-c.so (found version "3.2.5")
-- Found ZLIB: /usr/lib/libz.so (found version "1.3")
-- PyCXX found:
--   Headers:  /usr/local/include/python3.11
--   Sources:  /usr/ports/wip/freecad/work/FreeCAD-0.21.2-4887-g73cb5bafe0/src/CXX
--   Version:  7.1.8
-- Found OCC: /usr/local/include/OpenCASCADE (found version "7.8.1")
-- -- Found OCE/OpenCASCADE version: 7.8.1
[... 447 lines cut ...]
-- QtOpenGL:                    6.7.2
-- QtPrintSupport:              6.7.2
-- QtSvg:                       6.7.2
-- QtTest:                      6.7.2
-- QtUiTools:                   6.7.2
-- QtWidgets:                   6.7.2
-- QtXml:                       6.7.2
-- Rift:                        not enabled (BUILD_VR)
-- Shiboken:                    6.7.2 [/usr/local/include/shiboken6;/usr/local/include/python3.11]
-- SMESH:                       7.7.1.0 build internal
-- SPNAV:                       not found
-- SWIG:                        4.2.1
-- VTK:                         9.2.6
-- XercesC:                     3.2.5 [/usr/local/lib/libxerces-c.so] [/usr/local/include]
-- yaml-cpp:                    0.8.0
-- ZLIB:                        1.3

=================================================
Now run 'cmake --build /usr/ports/wip/freecad/work/.build' to build FreeCAD
=================================================

-- Configuring done (35.3s)
-- Generating done (2.5s)
-- Build files have been written to: /usr/ports/wip/freecad/work/.build
```

benj5378, 2024-09-01:
As far as I can see, it's due to 
https://github.com/FreeCAD/FreeCAD/blob/2a9cf9e8e6e62086948eb15e24d1e69a54f53d97/src/Gui/QuantitySpinBox.cpp#L444
Where `edit->displayText().size() - d->unitStr.size()` evaluates to a negative number; which is not allowed as function argument.

Looking at the precomming lines,
https://github.com/FreeCAD/FreeCAD/blob/2a9cf9e8e6e62086948eb15e24d1e69a54f53d97/src/Gui/QuantitySpinBox.cpp#L438-L444
it seems to me that `edit->displayText().size()` might evaluate to 0 thereby yielding the negative parameter value. Why only specific OS? Perhaps because the rendering or theme used on that OS gives this unexpected size.

@outpaddling, to do some further investigation, can you add
```
Base::Console().Message("QLineEdit, size=%i", edit->displayText().size());
Base::Console().Message("QLineEdit, text=%s", edit->displayText().text());
Base::Console().Message("d mystery variable, size=%i", d->unitStr().size());
Base::Console().Message("d mystery variable, text=%s", d->unitStr());
```
on the line before 444? Then run cause the crash again and supply the output we made.

outpaddling, 2024-09-01:
I think you're on the right track.  The patch did not compile, though.  I'm currently building with Qt6 in case that matters.

```
/usr/ports/wip/freecad/work/FreeCAD-0.21.2-4887-g73cb5bafe0/src/Gui/QuantitySpinBox.cpp:446:71: error: no member named 'text' in 'QString'
    Base::Console().Message("QLineEdit, text=%s", edit->displayText().text());
                                                  ~~~~~~~~~~~~~~~~~~~ ^
/usr/ports/wip/freecad/work/FreeCAD-0.21.2-4887-g73cb5bafe0/src/Gui/QuantitySpinBox.cpp:447:60: error: type 'QString' does not provide a call operator
    Base::Console().Message("d mystery variable, size=%i", d->unitStr().size());
                                                           ^~~~~~~~~~
/usr/ports/wip/freecad/work/FreeCAD-0.21.2-4887-g73cb5bafe0/src/Gui/QuantitySpinBox.cpp:448:60: error: type 'QString' does not provide a call operator
    Base::Console().Message("d mystery variable, text=%s", d->unitStr());
```

After changing the patch to the following:

```
--- src/Gui/QuantitySpinBox.cpp.orig    2024-09-01 17:20:00 UTC
+++ src/Gui/QuantitySpinBox.cpp
@@ -441,6 +441,12 @@ void QuantitySpinBox::updateEdit(const QString& text)
 
     edit->setText(text);
 
+    // Debug issue #16116
+    Base::Console().Message("QLineEdit, size=%i\n", edit->displayText().size());
+    Base::Console().Message("QLineEdit, text=%s\n", edit->displayText().toStdString().c_str());
+    Base::Console().Message("d mystery variable, size=%i\n", d->unitStr.size());
+    Base::Console().Message("d mystery variable, text=%s\n", d->unitStr.toStdString().c_str());
+
     cursor = qBound(0, cursor, edit->displayText().size() - d->unitStr.size());
     if (selsize > 0) {
         edit->setSelection(0, cursor);
```

I get this output:

```
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Coin error in cc_flwft_set_char_size(): FT_Set_Char_Size(.., 50, 50, ..) returned error code 23
QLineEdit, size=14
QLineEdit, text=-6' 11" - 3/4"
d mystery variable, size=2
d mystery variable, text=in
QLineEdit, size=14
QLineEdit, text=-6' 11" - 3/4"
d mystery variable, size=2
d mystery variable, text=in
QLineEdit, size=1
QLineEdit, text=0
d mystery variable, size=2
d mystery variable, text=in
ASSERT: "!(P(max) < P(min))" in file /usr/local/include/qt6/QtCore/qminmax.h, line 75                                                                           
Abort (core dumped)                                                             
```

Also, before I noticed that the rendering/theme hypothesis was discarded, I tried running under XCFE (instead of my usual Lumina).  It made no difference.

benj5378, 2024-09-01:
> Also, before I noticed that the rendering/theme hypothesis was discarded, I tried running under XCFE (instead of my usual Lumina). It made no difference.

Yeah, I thought `size()` in `edit->displayText().size()` referred to GUI size in pixels. Actually, it referred to the length of the string 🥲

benj5378, 2024-09-01:
Okay, gotta get to bed now. Will look at your output tomorrow.

outpaddling, 2024-09-02:
Adding the following hack prevents the crash:

```
--- src/Gui/QuantitySpinBox.cpp.orig    2024-09-01 17:20:00 UTC
+++ src/Gui/QuantitySpinBox.cpp
@@ -441,7 +441,14 @@ void QuantitySpinBox::updateEdit(const QString& text)
 
     edit->setText(text);
 
-    cursor = qBound(0, cursor, edit->displayText().size() - d->unitStr.size());
+    // Debug issue #16116
+    Base::Console().Message("QLineEdit, size=%i\n", edit->displayText().size());
+    Base::Console().Message("QLineEdit, text=%s\n", edit->displayText().toStdString().c_str());
+    Base::Console().Message("d mystery variable, size=%i\n", d->unitStr.size());
+    Base::Console().Message("d mystery variable, text=%s\n", d->unitStr.toStdString().c_str());
+
+    int size = edit->displayText().size() - d->unitStr.size();
+    cursor = qBound(0, cursor, size < 0 ? 0 : size);
     if (selsize > 0) {
         edit->setSelection(0, cursor);
     }
```

This, of course, is not a solution, just an ugly workaround.  We need to prevent the size from being negative in the first place.

I still get a crash when clicking "Done" on the start page (#16047), but I spent some time working on the model this morning and didn't experience any other problems.

benj5378, 2024-09-02:
@outpaddling, yeah, it's because the _ mm_ gets stripped from `QLineEdit edit`. It's value is given as the function argument `text`. If I try to delete the unit, it gets added again. Hmmm... referring to your previous traceback, could you try setting a breakpoint at `QuantitySpinBox::setValue` and step through the code to see where the unit is lost? Idk how lldb works, but with gdb, you can, as you might already know...
1. Start FreeCAD in GDB by `gdb FreeCAD` and `run`
2. Add the dimension and do everything except what is causing the crash
3. Set the the breakpoint with `b QuantitySpinBox::setValue` right before changing the dimension.
4. Print the function argument or other variables with the value with `p value` or `p text`. If it's a pointer just do `p *value`.
5. Step through the code with `n` and `s`. Use `n` to execute line by line of irrelevant code. Use `s` when executing lines for functions `QuantitySpinBox::updateText` and `QuantitySpinBox::updateEdit` to step into the function code and not just jump to the next line.

```cpp
    frame #8: 0x000000082762b619 libFreeCADGui.so`Gui::QuantitySpinBox::updateEdit(this=0x000001b6b7a6c600, text=0x0000000820e72da8) at QuantitySpinBox.cpp:444:14                                                                              
    frame #9: 0x000000082762bc73 libFreeCADGui.so`Gui::QuantitySpinBox::updateText(this=0x000001b6b7a6c600, quant=0x0000000820e72e38) at QuantitySpinBox.cpp:428:5
    frame #10: 0x000000082762c00f libFreeCADGui.so`Gui::QuantitySpinBox::setValue(this=0x000001b6b7a6c600, value=0x0000000820e72e38) at QuantitySpinBox.cpp:492:5
```

outpaddling, 2024-09-04:
I've poked at it a little and found it to be a cumbersome process, given that most of the variables are layered, complex data types.  FYI, lldb has basically the same commands as gdb, but it's disconnecting from the process shortly after starting the program.  ;-/

I've starting inserting more caveman-debug code to trace backward from the point of the crash.  It'll be a slow process as my time is limited right now.  I'm scrambling to finish building the balcony I designed with FreeCAD.  ;-)

luzpaz, 2024-09-07:
JFYI, some *BSD discussions happening in #16047, #16350 and this [forum thread](https://forum.freecad.org/viewtopic.php?t=80792)

outpaddling, 2024-09-12:
Here's a more complete backtrace, with debug info from Qt:

<details>
<summary>Backtrace</summary>
```
lldb -c FreeCAD.core /usr/local/bin/FreeCAD
(lldb) target create "/usr/local/bin/FreeCAD" --core "FreeCAD.core"
Core file '/home/bacon/FreeCAD.core' (x86_64) was loaded.
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGABRT
  * frame #0: 0x00000008411de86a libc.so.7`__sys_thr_kill + 10
    frame #1: 0x0000000841156f54 libc.so.7`__raise + 52
    frame #2: 0x0000000841207fc9 libc.so.7`abort + 73
    frame #3: 0x000000082831e0b3 libFreeCADGui.so`messageHandler(type=QtFatalMsg, context=0x0000000820dc0770, msg=0x0000000820dc0648) at Application.cpp:1931:13
    frame #4: 0x000000083f6e54d5 libQt6Core.so.6`qt_message(QtMsgType, QMessageLogContext const&, char const*, __va_list_tag*) at qlogging.cpp:2018:9
    frame #5: 0x000000083f6e547f libQt6Core.so.6`qt_message(msgType=QtFatalMsg, context=0x0000000820dc0770, msg=<unavailable>, ap=0x0000000820dc0740) at qlogging.cpp:371:5                                                                     
    frame #6: 0x000000083f6ea39a libQt6Core.so.6`QMessageLogger::fatal(this=<unavailable>, msg=<unavailable>) const at qlogging.cpp:889:5
    frame #7: 0x000000083f6e057e libQt6Core.so.6`qt_assert(assertion=<unavailable>, file=<unavailable>, line=<unavailable>) at qassert.cpp:68:14
    frame #8: 0x00000008286de373 libFreeCADGui.so`int const& qBound<int>(min=0x0000000820dc08ac, val=0x0000000820dc096c, max=0x0000000820dc08cc) at qminmax.h:46:5
    frame #9: 0x0000000828ed05ae libFreeCADGui.so`Gui::QuantitySpinBox::updateEdit(this=0x0000056744494600, text=0x0000000820dc0a08) at QuantitySpinBox.cpp:462:14                                                                              
    frame #10: 0x0000000828ed0c8b libFreeCADGui.so`Gui::QuantitySpinBox::updateText(this=0x0000056744494600, quant=0x0000000820dc0a98) at QuantitySpinBox.cpp:435:5                                                                             
    frame #11: 0x0000000828ed103f libFreeCADGui.so`Gui::QuantitySpinBox::setValue(this=0x0000056744494600, value=0x0000000820dc0a98) at QuantitySpinBox.cpp:512:5                                                                               
    frame #12: 0x0000000828b976e1 libFreeCADGui.so`Gui::PropertyEditor::PropertyUnitItem::setEditorData(this=0x00000567431e08c0, editor=0x0000056744494600, data=0x0000000820dc0b38) const at PropertyItem.cpp:1230:14
    frame #13: 0x0000000828beae67 libFreeCADGui.so`Gui::PropertyEditor::PropertyItemDelegate::setEditorData(this=0x000005671e902c00, editor=0x0000056744494600, index=0x0000000820dc0bb0) const at PropertyItemDelegate.cpp:236:20
    frame #14: 0x000000083ae267b5 libQt6Widgets.so.6`QAbstractItemViewPrivate::updateEditorData(this=0x000005671edd4600, tl=0x0000000820dc0e40, br=0x0000000820dc0e28) at qabstractitemview.cpp:4510:23
    frame #15: 0x000000083ae294d3 libQt6Widgets.so.6`QAbstractItemView::dataChanged(this=0x000005671e7b2800, topLeft=0x0000000820dc0e40, bottomRight=0x0000000820dc0e28, roles=<unavailable>) at qabstractitemview.cpp:3423:12
    frame #16: 0x000000083f7b5f84 libQt6Core.so.6`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(this=0x000005671e6fcc80, r=0x000005671e7b2800, a=0x0000000820dc0d70) at qobjectdefs_impl.h:469:51
    frame #17: 0x000000083f7b5f6e libQt6Core.so.6`void doActivate<false>(sender=0x000005671ee04540, signal_index=3, argv=0x0000000820dc0d70) at qobject.cpp:4086:26
    frame #18: 0x000000083f99b15d libQt6Core.so.6`QAbstractItemModel::dataChanged(this=<unavailable>, _t1=<unavailable>, _t2=<unavailable>, _t3=<unavailable>) at moc_qabstractitemmodel.cpp:1022:5
    frame #19: 0x0000000828bf0952 libFreeCADGui.so`Gui::PropertyEditor::PropertyModel::updateChildren(this=0x000005671ee04540, item=0x00000567431e2a00, column=1, parent=0x0000000820dc0ef8) at PropertyModel.cpp:539:16
    frame #20: 0x0000000828bf080b libFreeCADGui.so`Gui::PropertyEditor::PropertyModel::updateProperty(this=0x000005671ee04540, prop=0x0000056742feab40) at PropertyModel.cpp:484:5                                                              
    frame #21: 0x0000000828b816e6 libFreeCADGui.so`Gui::PropertyEditor::PropertyEditor::updateProperty(this=0x000005671e7b2800, prop=0x0000056742feab40) at PropertyEditor.cpp:662:24                                                           
    frame #22: 0x000000082895138d libFreeCADGui.so`Gui::PropertyView::slotChangePropertyData(this=0x000005671ed86a00, prop=0x0000056742feab40) at PropertyView.cpp:218:29                                                                       
    frame #23: 0x0000000828955be1 libFreeCADGui.so`decltype(*std::declval<Gui::PropertyView*&>().*std::declval<void (Gui::PropertyView::*&)(App::Property const&)>()(std::declval<App::Property const&>())) std::__1::__invoke[abi:v160006]<void (Gui::PropertyView::*&)(App::Property const&), Gui::PropertyView*&, App::Property const&, void>(__f=0x000005671e902660, __a0=0x000005671e902670, __args=0x0000056742feab40) at invoke.h:359:23
    frame #24: 0x0000000828955b6a libFreeCADGui.so`std::__1::__bind_return<void (Gui::PropertyView::*)(App::Property const&), std::__1::tuple<Gui::PropertyView*, std::__1::placeholders::__ph<2>>, std::__1::tuple<App::DocumentObject const&, App::Property const&>, __is_valid_bind_return<void (Gui::PropertyView::*)(App::Property const&), std::__1::tuple<Gui::PropertyView*, std::__1::placeholders::__ph<2>>, std::__1::tuple<App::D [... cut ...]
    frame #94: 0x00000008283220fb libFreeCADGui.so`(anonymous namespace)::tryRunEventLoop(mainApp=0x0000000820dc5e00) at Application.cpp:2172:5
    frame #95: 0x000000082831f383 libFreeCADGui.so`(anonymous namespace)::runEventLoop(mainApp=0x0000000820dc5e00) at Application.cpp:2192:9
    frame #96: 0x000000082831ea1e libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2269:5
    frame #97: 0x000000000022d086 FreeCAD`main(argc=1, argv=0x0000000820dc6bb8) at MainGui.cpp:301:13
    frame #98: 0x000000084112bafa libc.so.7`__libc_start1 + 298
    frame #99: 0x000000000022adb0 FreeCAD`_start at crt1_s.S:83
```
</details>

outpaddling, 2024-10-27:
I'm no longer seeing a crash in RC2, so closing the issue.  Thanks for the efforts!

closed 2024-10-27 by outpaddling

luzpaz, 2024-10-28:
@outpaddling can you paste the RC2 About info (for posterity), TIA