FreeCAD/FreeCAD #16121: [TechDraw] Surface finish symbols to small

benj5378, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Requirements to do surface finish symbols: keyboard, mouse and **microscope**.

Would anyone mind if I changed the buttons to enlarge the icons? @WandererFan 

[image]

### Full version info

```shell
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38514 (Git)
Build type: Debug
Branch: centeredDimensionLabel
Hash: 85ac7a4b227d724baf3c3b4dc33de42595e43721
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Help 1.0.3
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-28:
For me, something like the buttons being 50x50 ish pixels, stroke-width increased inline with the size and it would be really nice if the pushbutton icons and the `<PySide2.QtWidgets.QGraphicsView(0x5a8d47a4d180, name="graphicsView")` were all dark stylesheet aware like the _Annotation_ Preferences are already, that would be a polished UI IMHO.

FEA-eng, 2024-08-28:
Global issue for this tool: #12719

benj5378, 2024-09-04:
#16147

closed 2024-09-04 by benj5378