FreeCAD/FreeCAD #16125: \Mod\Drawing\  files are not in CMakeLists.txt

aiksiongkoh, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The following files are not in CMakeLists.txt
src\Mod\Drawing\App\
FeatureProjection.h
FeatureViewPart.h
FeatureViewSpreadsheet.h
...

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD/tree/main/src
today
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-08-29:
Are we still building the old Drawing module?

luzpaz, 2024-08-29:
IIRC there is Drawing code still used somewhere..I can't recall where ATM

WandererFan, 2024-08-30:
I though all the remaining Draw functions (Project shape, etc) had been moved to TechDraw.

luzpaz, 2024-08-30:
@WandererFan that may be true, but what was the decision to leave remaining files then ?

wwmayer, 2024-11-09:
All the mentioned files are part of https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Drawing/App/CMakeLists.txt

xtemp09, 2025-04-05:
I think it will be better if the issue be closed.

`FeatureProjection.h` and `FeatureViewPart.h` were added with the first commit. `FeatureViewSpreadsheet.h` was added as soon as it was created. You may want to take a look at this commit, 85fdc8c76bd504b0cd70e02ee16c1cb75faa8214.

closed 2025-04-05 by luzpaz