FreeCAD/FreeCAD #16127: Sketcher - Line tool constrains do not work upon creation.

GAZ082, 2024-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you create a closed shape with the line tool OR use the line tool in combination with other shape, the created coincident constrains are not recognized. You need to actually move a point so the constrains gets refreshed (i guess) and then works. I have to stress, this also happens if you combine the line tool with another tool in the sketcher.

https://github.com/user-attachments/assets/427ead3c-cb47-4252-9c0f-54d88813749e

**MACRO:**
[line_Test.FCMacro.Zip](https://github.com/user-attachments/files/16790150/line_Test.FCMacro.Zip)

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves.backup1713713759.0032098 0.6.29 (Disabled)
  * QuickMeasure 2022.10.28
  * ExtremeProDark 2.9.0
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * CfdOF.backup1710694906.6805208 1.24.9 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * Curves.backup1714093860.7275996 0.6.34 (Disabled)
  * OpenTheme 2024.8.17
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Assembly4.backup1710902627.637761 0.50.9 (Disabled)
  * Defeaturing.backup1710696629.0288966 1.2.2 (Disabled)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GAZ082, 2024-08-29:
Highly suspicious. No  thanks.

maxwxyz, 2024-08-29:
@GAZ082 is this also the case in the stable 0.21.2 version?

kaiwas, 2024-08-29:
Not confirmed

https://github.com/user-attachments/assets/420b96f7-a1e2-4d2a-b7e2-3758eeaa6d59

```

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.8.17
  * PitchedRoof

```

FEA-eng, 2024-08-29:
Can't confirm either.

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

luzpaz, 2024-08-29:
An aside, troubleshooting this issue further justifies #15710

FEA-eng, 2024-08-29:
Did you change something in Advanced solver control?

[image]

This section is not visible by default. Please try restoring the defaults within it.

GAZ082, 2024-08-29:
> @GAZ082 is this also the case in the stable 0.21.2 version?

Sadly cant verify. Both the APPIMAGE and my local installation do not like my system:
[image]

GAZ082, 2024-08-29:
> Did you change something in Advanced solver control?
> 
> ![advanced](https://private-user-images.githubusercontent.com/59876896/362659764-e46529fe-b9ea-4306-9ea0-68f96a230008.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ5MzM1ODAsIm5iZiI6MTcyNDkzMzI4MCwicGF0aCI6Ii81OTg3Njg5Ni8zNjI2NTk3NjQtZTQ2NTI5ZmUtYjllYS00MzA2LTllYTAtNjhmOTZhMjMwMDA4LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI5VDEyMDgwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTMxYWYyZjc5OWEyODQyZGQ1YjBlNzExNDgzZjIxMmRmMmI4ZjQ4NjViYWIwYzVmN2NiNjk5MmFjYmQxZGYwYmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.O0hUnX_Ub-uJ_U11rwTCSlQDbHHgA04JsgHvr52zAUM)
> 
> This section is not visible by default. Please try restoring the defaults within it.

Reverted to defaults, no luck. I even restarted FC just in case.

GAZ082, 2024-08-29:
Guys, try in the XZ pane. XY seems fine! @kaiwas @FEA-eng

GAZ082, 2024-08-29:
This is one of those bugs  that are sneaky... check this.

Uploading Video_2024-08-29_09-41-11.mp4…

GAZ082, 2024-08-29:
I can confirm after multiple tests that the issue is with the XZ pane.

Syres916, 2024-08-29:
Confirmed, what's even more bizarre if I create the sketch in pure Sketcher (no Part Design involved) it can be extruded in the Part Wb fine. I thought both used the same code in _Part::FaceMaker::addTopoShape_ but this maybe proves they don't. I'll check if this was a recent regression first.

kaiwas, 2024-08-29:
> for me:

+1

FEA-eng, 2024-08-29:
@GAZ082 Can you create a backup of your config file and then delete it from the original location to have fresh config when starting FreeCAD?

I will also try on Ubuntu just in case.

Syres916, 2024-08-29:
This is not a recent regression, running the macro using:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0f9259c)
Hash: 0f9259cda103ae1824ac16c68ac9b4a0d54b05fc
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

shows this has been hiding away for a long time. Even just adding an angle constraint between the lower edge of the triangle and the X axis therefore not touching any vertexes fixes the _Wire Not Closed_ so (for me) it's not going to be an easy fix. If it was failing when fully constrained then I would say the priority would need to be raised but as it only applies when the geometry is 'free' then I'd say it's a good one to fix but not a show stopper IMHO.

FEA-eng, 2024-08-29:
> This is not a recent regression, running the macro using:

Did you modify this macro to run it? I get:

```
15:56:48  Traceback (most recent call last):
  File "C:/Users/username/Desktop/line_Test.FCMacro", line 34, in <module>
    lastGeoId = len(ActiveSketch.Geometry)
<class 'NameError'>: name 'ActiveSketch' is not defined
```

when trying.

Syres916, 2024-08-29:
You have to be in Sketch Edit Mode before running the macro

luzpaz, 2024-08-29:
An aside: 
For years I've been seeing how the configs get corrupted and we don't really fully know why.  It would be beneficial if we ran some background process that just watched config history to see wth changes create these anomalies.

luzpaz, 2024-09-09:
@GAZ082 have you been able to reproduce ?

GAZ082, 2024-09-09:
> @GAZ082 have you been able to reproduce ?

Hi @luzpaz . No. Clearing the configs "worked".

JohnOCFII, 2024-09-12:
An easy "Safe Mode" option to re-start without using existing configuration files would be a nice troubleshooting tool to add to FreeCAD.

luzpaz, 2024-09-12:
> An easy "Safe Mode" option to re-start without using existing configuration files would be a nice troubleshooting tool to add to FreeCAD.

#6106
https://forum.freecad.org/viewtopic.php?t=50342

closed 2025-07-31 by github-actions[bot]