FreeCAD/FreeCAD #16132: [Feature Request] [Draft] Snap Extension should also work for curved edges

Roy-043, 2024-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently Snap Extension only works for straight edges. It should also work for curved edges. 'Virtual' intersections with those extended edges should then also be supported. For example the intersection between an extended arc and a Snap Ortho line.

### Full version info

```shell
V0.22
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-29:
Related forum topic:
https://forum.freecad.org/viewtopic.php?t=49904

gianlucafiore, 2024-09-14:
Could something be done so that the extension restrictions are not so heavy on memory? When I use a drawing imported in DXF with dimensions and annotations, it is almost impossible for me to use the tool

[image]