FreeCAD/FreeCAD #16135: Cannot select multiple faces/edges when Body contains a boolean from a shapebinder

maxwxyz, 2024-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a solid from a shapebinder in PartDesign I cannot select multiple faces or multiple lines with CTRL + Click. Is this a known limitation @wwmayer ?
[image]

When selecting the original solid (base for the shapebinder) there is no issue in selecting multiple elements.
Also it is possible to select multiple features of the shapebinder itself but not after I've used it in a boolean operation.

Demo file:
[Demo.zip](https://github.com/user-attachments/files/16799368/Demo.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-29:
Adding a feature to the boolean (e.g. a fillet) allows a multi selection again.
@bgbsww could this be related to TNP changes (I have no chance to test on stable currently)?

wwmayer, 2024-08-29:
If you open the selection view you will see that the multi-selection actually works but additionally selected sub-element are not rendered correctly.

In v0.21 this works correctly. So, it's regression in v0.22

maxwxyz, 2024-08-29:
Thanks! Any idea what caused this?

bgbsww, 2024-08-29:
Yeah, selection of faces in a binder seems in general to not highlight
correctly.  I'm not sure it's TNP or not, but I'll take a more in depth
look when I get home tomorrow if it hasn't been nailed by then.

bgbsww, 2024-08-30:
Hmm.  Does changing the Use Binder Style property in View to false make this go away, or am I looking at the wrong thing?  If that's it, tracking down the history of changes to this might lead to a path - and way away from TNP.

maxwxyz, 2024-08-30:
@wwmayer @bgbsww it works normally in thy stable 0.21.2 version.
In 0.22dev changing the Use Binder Style property in View to false makes no difference, it is always one element displayed as selected. It is also no issue when selecting the binder itself, only after it is used in a boolean operation.

[image]

maxwxyz, 2024-08-30:
@bgbsww these all as regression to 0.21.2?

bgbsww, 2024-08-30:
Actually, no.  Looks like the dialog duplication and broken remove button
are all also in 21.2,  but the selection works.

bgbsww, 2024-08-31:
More:  this seems to be about what the view property Display is set to for the Boolean:  If it is Result, then the behavior of only showing the first selection happens.  If it is tools, selection gets the source objects and multiselection works properly.

In addition, it appears to be possible to be in Result display mode while Display is set to tools in some initial cases - maybe a missing recompute in my test code - and this will also show the bug until refreshed.

maxwxyz, 2024-08-31:
@bgbsww recompute does not solve the issue. Changing it to tools just displays the binder and selection works as if I only have the binder without the boolean object. But the display result is the desired outcome which is needed so it should work there. Looking through the PRs I cannot find an obvious one by the title which lead to this regression.

bgbsww, 2024-08-31:
Actually,  I've found something:  If you set a breakpoint on SelectionSingleton::addSelection and try selecting vs ctrl-selecting a face in a boolean, the former works and the latter fails.  And the former has a full mapped name to the pSubname, and the latter has  a classic name.  So it is a TNP breakage, and it is about how the add to selection as opposed to the initial selection works.

bgbsww, 2024-08-31:
Actually, that difference wasn't actually what was breaking the selection in booleans, and may not have been necessary although I'm keeping it.  The real issue on the booleans was that the ctrlDown path had no code in it at all to update the visuals.

I have that fixed and will now look at the shapebinder specific case as soon as I finish updating to latest main to avoid a conflict on a FC_USE_TNP_FIX section.

PR #16203 (2024-09-02): Toponaming: Fix ctrl selection issues by looking up element names and…
… setting up detail path.  Fix #16135.  Fix #15882.
closed 2024-09-02 by yorikvanhavre