FreeCAD/FreeCAD #16146: [Feature] Sketcher taper constraint

Anthony-Gaudino, 2024-08-29:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I propose the addition of a taper constraint in the Sketcher.

Sometimes I encounter a blueprint that has something like `.09 INCLUDED TPI BASIC` which means a .09" taper per inch.
This can be calculated to get an angle but it would be easier to have a way to set the taper as a constraint. Usually the taper is results in a very small angle which is hard to see in FreeCAD. Also it would more exactly match what the blueprint is asking for.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * Launcher
  * Nodes 0.1.36
  * InventorLoader 1.4.0
  * Assembly4.backup1719958896.85335 0.50.12 (Disabled)
  * Render.backup1719959316.120863 2024.6.5 (Disabled)
  * woodworking 0.21.2.33771
  * ThreadProfile.backup1719959316.6397839 1.90.0 (Disabled)
  * MeshRemodel.backup1719959315.871541 1.9.23 (Disabled)
  * Silk 0.1.5
  * Defeaturing 1.2.2
  * Pyramids-and-Polyhedrons
  * Curves.backup1719958901.528519 0.6.36 (Disabled)
  * fasteners.backup1719958901.808955 0.5.20 (Disabled)
  * nurbs
  * sheetmetal.backup1719959316.3711638 0.4.21 (Disabled)
  * A2plus.backup1719958896.309429 0.4.64a (Disabled)
  * Manipulator 1.5.7
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * toSketch 1.0.1
  * Curves 0.6.45
  * Assembly4 0.50.15
  * Assembly3 0.12.2
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-29:
@Anthony-Gaudino can you mockup your proposal perhaps ?

Roy-043, 2024-08-30:
You can use an expression for the angle: `atan(0.09)`. Have you tried that?

pierreporte, 2024-08-30:
FreeCAD already allows to enter values with specified unit. It may be sufficient to provide a TPI angle unit, as well as a percent or per mille angle units. There could also be a clever calculation when an entered angle is like “1 mm/m”. Both can work with the current UI.