FreeCAD/FreeCAD #16148: Part Design: Body > Placement - Reset + Expressions and Undo bugs

samohd, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are two or three bugs here.

In Part Design: Body > Model panel > Data > Placement, double-click ... to edit Placement via Task Panel. Set Translation > Y > Expression to 10mm. Object moves as expected in view. Click Reset. The Y value changes from 10mm to 0 and the view reflects this. But, the Expression dot is still blue. Click it to edit Expression. It shows 10mm. Click OK. Y still shows 0mm. Click to edit the Expression again. Change it to 15mm. Click OK. Y now shows 15mm. Click Reset. Y Shows 0mm and view changes to reflect that. Click OK to exit Placement panel. View shows the position jump back to 15mm.

Double-click ... to edit Placement via Task Panel again. Change Y's Expression to 20mm. Click OK. Click OK again to exit Placement panel. View shows the position jump  to 20mm. Press Ctrl-Z to undo. Nothing happens. Ctrl-Z again. View changes. Double-click ... to edit Placement via Task Panel again. Y shows 15mm. Edit the Expression... it shows 20mm.

Good fun eh.

These issue may exist with other parameters than Y. Y just happens to be the one I was using when I discovered the issue. May happen in other similar panels too. But, this is all I've looked at.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-09:
@samohd would you be so kind to create a screencast for this so devs grok the issue quickly ? TIA

samohd, 2024-09-09:
Actually for this kind of bug, it's better to follow the steps I detailed to produce the issue. As the developer can more easily track what parts of code are in play.

luzpaz, 2025-06-04:
This bug got no traction because it was hard to read, imho. I added some markup for better readability.