FreeCAD/FreeCAD #16149: [Feature Request] TechDraw: Any property linkable to autofill fields by Category and Label

Roundwolf, 2024-08-30:
It would be helpful if it is possible to define the source for the autofill information by yourself. For example, like you refer to the "Created By" field or the "Company" field in the base properties, you could add your own property to the base properties and could refer to it in your drawing template by the label of the property. This would also allow you to all the other properties in BaseProperties. I hope that wasn't too confusing. 

Furthermore, the material should automatically fill in with the upcoming material workbench.

_Originally posted by @Roundwolf in https://github.com/FreeCAD/FreeCAD/issues/5903#issuecomment-2317637483_

luzpaz, 2024-09-09:
cc @pavltom do you have time to consider this feature ?

pavltom, 2024-09-10:
Hello @luzpaz, unfortunately this autumn is pretty busy for me. I will try to find some spare time, but chances are low. Thanks for understanding.