FreeCAD/FreeCAD #16150: Stylesheet: Change to classic does not set "No stylesheet" / Line between views.

maxwxyz, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

@obelisk79 as  requested on Discord, here the issue:

When changing to the Classic theme from a different theme, the stylesheet is not reset to "no stylesheet".
[image]
[image]

When changing to no stylesheet, there is a line between the tree view and property view when using transparent overlays, only when they are set as combined (=default) in the preferences. When set to separate, there is no line between the views.
@kadet1090 FYI

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

obelisk79, 2024-08-30:
So, conducting a local test, when I set FreeCAD Classic via preferences, it properly resets FreeCAD to no stylesheet. I'm actually unable to recreate the issue you are seeing on my Linux machine.

maxwxyz, 2024-08-30:
@MisterMakerNL can you confirm on Windows?

Syres916, 2024-08-30:
@obelisk79 try cutting the post.FCMacro down to just the first 8 lines.

obelisk79, 2024-08-30:
> @obelisk79 try cutting the post.FCMacro down to just the first 8 lines.

Thanks for the hint. I actually found removing the macro entirely provided satisfactory results. If others are able to confirm, I'll submit a PR to remove it later today.

JohnOCFII, 2024-08-30:
> @JohnOCFII could you confirm this in the latest weekly?

I find testing the stylesheet issues that are dependent on HOW you change to the stylesheet to be very difficult to test properly.  I never know if I'm getting to the situation in the required fashion.  I was previously set to OpenLight, and used the Preference window to switch to FreeCAD Classic.  Nothing crazy wrong to me, other than (standard?) tiny little separation bar (circled)  on the left that most won't see.

Overall, I do think the combination of transparency and overlay toggles - while powerful, can be confusing.

[image]

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.17
  * fasteners 0.5.27
```

maxwxyz, 2024-08-30:
@JohnOCFII when switching to classic, what does the stylesheet preference say under Display UI? "No stylesheet"?
@MisterMakerNL in above screenshot the header bars in the task dialog are also missing (https://github.com/FreeCAD/FreeCAD/issues/15802)

MisterMakerNL, 2024-08-30:
The headers are from the stylesheet, which pulls the colors from the accent colors. So seems like the stylesheet is still there but the accent colors are removed.

JohnOCFII, 2024-08-30:
> @JohnOCFII when switching to classic, what does the stylesheet preference say under Display UI? "No stylesheet"?

When switching from OpenLight to FreeCAD Classic under `Preferences --> General` (and hitting `Apply` then `OK`) if I switch to `Preferences --> Display --> UI` it shows stylesheet as `OpenLight.`
[image]

maxwxyz, 2024-08-30:
ok thanks. @obelisk79 so confirmed for Windows and macOS

maxwxyz, 2024-08-31:
Should be resolved by https://github.com/FreeCAD/FreeCAD/pull/16169

closed 2024-08-31 by maxwxyz