FreeCAD/FreeCAD #16152: Add possibility to add several tree views (current tree, historical tree, BOM tree, Body tree, IFC tree)

Mannshoch, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The tree view is not well organized. I assume that a Workbench developer has to decide for himself how the structure looks like and - at least not known to me - there is no possibility to display the displayed structure according to other needs.

In my experience, there are the following display types that are important when designing multi-part geometries in particular.
- History  / Assembly view
- current view / Project structure
- Body view / BOM
- BOM

Description of the display types:

## History view:
May default view in Part-Design
The history based list is especially needed for **PartDesign** modeling. It effectively reflects the script that runs from start to finish to display the CAD data.
By cleverly arranging functions in the right place in the script, you can:
- create a geometry with a few intelligently used functions that is easy to understand for third parties and easy to modify
- Access geometries and information that are no longer available later on.
- Favors an organic and evolutionary development of a part.
- Project structure appear inside the History tree. Mostly at the top of the tree.
- find every feature and function that add Information. Some information could be collapsed inside other objects.
- Assembly mode:
  - You have a tree that is split into two section. Assembly and History. The first section is Assembly and the second is history.
   - The Assembly section only appear if more than one independent Bodies (not if a Feature modify more than one body together) are available.
      --> I call such independent bodies "Parts"
  -  If you create a sketch reference on one body that has constraints to a second independent body, the sketch is in the history tree section.
  -  If you create a feature that modify more than one body (but not all) at once, all affected bodies are moved into an automatic created Part.
  -  Every so called "Part" could also be an other linked FreeCAD File. If you wish you could move a "Part" into it's own FreeCAD file or you could seamless integrate an external FreeCAD file into the current working FreeCAD file.
  -  It should be possible that a Sketch has constrains from other bodies. If a body move the sketch get updated.
    - --> It should be possible that such moved bodies update the sketch and this update force an update of the assembly mates which may affect again the sketch. To prevent an infinite loop check on this is needed.

## Current view / Project structure:
Project structure-based design is particularly important in situations where the designed elements ideally do not require other elements to be present. But there is still a relationship - usually in terms of position and alignment.
An example of this is an assembly or the Dodo Workbench, which positions parts in relation to each other. A history in this context would not be the order in which you have placed elements, but rather the script that performs the calculation at the end and automatically. In this case, however, it makes little sense to give the user the option of completely modifying the calculation process.

## BOM:
Currently BOM  is only a [spreadsheet](https://ondsel.com/blog/introducing-bom-tool/) but Boom is more.
I propose that a Bom Tree view shows a selected BOM-structure along with Items not listed there. 
In this tree view you could
- add/remove to BOM
- Group bodies
- reorder or regroup by drag and drop 
- set a group or assembly that it should be shown as Part
- select a Group or part and fill missing attributes

The BOM is in fact not more than an enhanced body view with additional possibilities like re-group bodies and mark them as sub-assembly. Or mark a Linked Assembly as absorbed to get them into the same BOM level.
More Important, you could set up properties like descriptions, article numbers, materials, BOM-Number per body or per group. (Information are stored in the history at the same location where a body the first time was created.

## Body view:
A body view shows all bodies in a File. They get grouped by e.g. Projects. If some are identical in the same project they get automatically grouped. Inside this list you get the opportunity to manually group bodies.
More enhanced, in context of assemblies does a body list not only shows bodies but also different projects, linked files and also their bodies.
A selection set that allow to highlight a set of bodies or use it to repeatedly Export selected Bodies

# My Enhancement Request:
Please create possibility to change the Tree view according the described Cases.
- History view is auto-generated. And a Rollback bar is needed. User could move e.g. sketches to earlier position as long by the sketch used references do exist at that time.
- Project structure could represent the current view and is controlled by the Workbenches.
- Body view is auto-generated, could be modified by user and reset if needed
- BOM view is auto-generated, could be modified by user and reset if needed

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git) Snap 1120
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Installed mods: 
  * PieMenu 1.8.0
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-30:
@kadet1090 could be combined for Part/PartDesign merge ideas

kadet1090, 2024-08-30:
While I like idea described in this feature (kudos to author!) I don't think that we should combine two such big projects together. This way we won't finish any of them. That said I can try to lay grounds for something like that but this seems like enormous task on its own.

pierreporte, 2024-08-30:
Ability to define multiple tree views for different types of work is necessary if a decent file structure is implemented. Drawings would have a specific structure for example, and BIM projects as well.

The proposed changes are large and depend on how we want FreeCAD to work. We already discussed this a little bit for the Part/PartDesign merge