FreeCAD/FreeCAD #16157: Toolbars are hard to navigate and hard to design properly

kadet1090, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD has a lot of tools and functions that must be accessible to users. Currently users can access them via:
1. Toolbars - which have icons and are easily accessible
2. Menu Entries - which need a lot of clicks
3. Shourtcuts - which user needs to remember

While toolbars are generally fine they lack ability to create more advanced users interfaces as every icon is the same size, some of them are hard to understand and some of them do not to be accessible in one click anyway. We need ability to design and implement better toolbar like structures similar to what Ribbon UI offers.

We need something that:
1. gives us design flexibility of ribbon ui
2. We works with current toolbars as we won't be able to rework everything 

The observation is that the sole thing that brings benefits of ribbon is that the action groups (which are basically toolbars) are higher and so can contain more variety of button sizes, labels etc. Thanks to that we can have big buttons for important stuff, smaller buttons with labels for less important stuff, grids of icons for stuff that is self explanatory etc. However with great power comes great responsibility - we will have more options but we will have to use it wisely. There is no way to provide automagically a good ribbon style UI based solely on what we have now - basically everything will have to be designed and not put randomly.

So if we somehow implement solution that allows us to create toolbars spanning across two rows and with custom layout we basically can have something that brings all of benefits of ribbons while still being backwards compatible. Workbenches that will provide two-row layouts for toolbars will be able to have ribbon-like experience, while for all others it will stay the same. This will allow us to transition smoothly and keep the old UI for those who like it. We may be even able to utilize some parts of existing libraries (that said - we will have to put them in-tree them in to have full control over the code) to ease the job.

What remains to be done:
1. Figure out how to implement toolbar rowspan in Qt
2. Find most edge cases and describe them
3. Implement layouts for all builtin toolbars

[image]

### Full version info

```shell
0.22dev / 1.0
```

### Subproject(s) affected?

None

### Anything else?

There are addons like: https://github.com/geolta/FreeCAD-Ribbon -  but such feature must be implemented in core and ribbon toolbars must be properly designed to work properly.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-09-01:
Discussion #16156 is about introducing short name for commands to be displayed in the ribbon. Do you plan to implement this with your own prototype?

MisterMakerNL, 2024-11-18:
This is an interesting one. you basically want more dynamic toolbars.
Would be nice if the user could get into a toolbar edit mode where it is possible to reorder icons and change icon size.
And then the option to save these settings.
We can also make a good default layout.

kadet1090, 2024-11-18:
> Would be nice if the user could get into a toolbar edit mode where it is possible to reorder icons and change icon size.
> And then the option to save these settings.

If I find time to implement it I plan for it to have such feature. I may apply for a grant to do this after transparent previews are implemented.