FreeCAD/FreeCAD #16158: Transparenz und Farbe

Solarjojo, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Wenn man an einem Körper etwas ändert geht die Farbe und die Transparenz verloren. s.Video

https://github.com/user-attachments/assets/da7e2b15-a28e-440a-98ea-83bf6365ced3

### Full version info

```shell
Wenn man an einem Körper etwas ändert geht die Farbe und die Transparenz verloren. s.Video
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-30:
@Solarjojo please translate ticket to english. Also please add your full [About](https://wiki.freecad.org/About) info + zip up a compressed file and drag it into github here to attach so it can be tested. thank you.

@maxwxyz is this a dupe ?

NomAnor, 2024-08-30:
His issue is that when he extrudes the sketch in his video, the color and transparency settings revert to their defaults.
The expactation is that after extruding, the result keeps the color and transparency of the previous feature.

I experienced the same. There were always some issues in Part Design with the color only set at the feature and not body level. But I never looked into it.

maxwxyz, 2024-08-30:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/15503

closed 2024-08-30 by maxwxyz

Solarjojo, 2024-08-30:
Andre explained it korrekt.