FreeCAD/FreeCAD #16159: About how to translate or not translate Privacy Policy and FCP Files

Reqrefusion, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The website pulls these two files directly from this repo, and since these files belong to the repo, I found it right to open this topic here. The website is completely translatable into other languages. @yorikvanhavre has previously stated that he is very sensitive about this issue. However, these documents are untranslatable. I foresee two ways: Since these documents are official, they should only be in English, and for this to be the case, there should be a translation text first, and as a second way, these documents should be opened for translation.

### Full version info

```shell
All Version
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-09-02:
These two documents are markdown files inside the FreeCAD source code. Indeed they are sensitive docs that should have a central, "official" position that affects the whole FreeCAD ecosystem, so having them in the source code seems the correct move to me.

Maybe we could have the crowdin script pick them and send them to crowdin. That should be easy. Then the crowdin update script on the homepage could unpack those translations. So the source would stay in the source code only, but the translations would be on the website only.

Last thing to check, make sure crowdin handles these markdown files correctly. We could do a test...

Reqrefusion, 2024-09-02:
> These two documents are markdown files inside the FreeCAD source code. Indeed they are sensitive docs that should have a central, "official" position that affects the whole FreeCAD ecosystem, so having them in the source code seems the correct move to me.

Considering the official nature of these documents, it does not seem right to me to use them without a directly accepted translation. Rather than a translation for the website, it would be more appropriate to have an text explaining the official nature of these documents and that they can only be in English. This is generally how it is done in international agreements. I was going to explain how these are done later on here, but I deleted it because it could be confusing.

yorikvanhavre, 2024-09-03:
They are official, but they are not like contracts. They don't engage responsibility, or that kind of thing, so I would assume user-made translations would be fine here, even if there is slightly different meaning, it would probably not be a big issue