FreeCAD/FreeCAD #16164: Select feature in tree view which created the selection in the 3D view (Sync selection of 3D element and feature)

maxwxyz, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This would be especially useful in PartDesign and is implemented in commercial CAD software:
When selecting an element of the model in the 3D view, e.g. a fillet, this feature should be selected in the tree view automatically. 

This would in the future also allow to edit this fillet feature by double clicking the fillet face in the 3D view without the need to look for the correct feature in the tree view / history.

Currently, a selection of an element in the 3D view always selects the last feature of a Part Design body, which also leads to issues like: 
- #11132 

Maybe the same could be done the other way around: select/hover a feature in the tree view and the corresponding geometry elements are highlighted, like mentioned in:
- https://github.com/FreeCAD/FreeCAD/issues/11215

This is the current behavior, selecting the filet selects the last feature - a chamfer:
[image]

@bgbsww this should be possible with the TNP-fixes, is this assumption correct? Would each element know where it came from, otherwise fall back to the latest it can, otherwise select the body or last feature.
@pierreporte FYI

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

PartDesign

### Anything else?

Maybe related https://github.com/FreeCAD/FreeCAD/issues/11215

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-08-31:
So, I agree with the idea here.  I don't see any TNP tie in:  this is "just" improving the selection to set the tree selection to match the first feature (defining) that has the selected subelement instead of the last feature (which is the definitive reference to that subelement in a body).  That seems to be selection work independent of if TNP is there or not - could always have been done.

It does reflect a more user-centric approach to how the UI interacts with a Body, and I suspect needs close analysis of all the side effects both good and bad.

maxwxyz, 2024-08-31:
@FreeCAD/design-working-group what is your opinion?

pierreporte, 2024-08-31:
@maxwxyz It’s as necessary as #11429.

maxwxyz, 2024-08-31:
in the RT link branch this is somewhat implemented the other way around: hovering the feature in the tree view, displays the face/geometry which was created by that feature. It does not work as this issue / feature request which would also be the other way around (select in 3D view and display the tree view feature).

https://github.com/user-attachments/assets/b1322f48-b8f4-4d0c-9468-63d42c5455eb

obelisk79, 2024-08-31:
I think both should be implemented.

bgbsww, 2024-08-31:
@FreeCAD/design-working-group Can I suggest that we get a unified opinion on how preview and select should ideally work and then roll all the associated bug/feature requests into one thing?  I'm worried that we play whack-a-mole here without shooting at a single target.  The selection work itself looks to be a bit fiddly, but localized and not difficult.  Factoring in whether the LS3 behavior, especially around preview would be wanted seems helpful.

maxwxyz, 2024-08-31:
agreed.
- https://github.com/FreeCAD/FreeCAD/issues/11214
- https://github.com/FreeCAD/FreeCAD/issues/11215

should be considered as well

NoxAstrum1, 2025-05-06:
This is a huge deal for me and would probably be my #1 need. I would want it to work in both directions. In a complicated part or assembly, it becomes very difficult to keep track of what's what.

maxwxyz, 2025-05-07:
I think @kadet1090 is working on this with transparent preview