FreeCAD/FreeCAD #16166: Draft: Polar array, original is missing

theosib, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached array, please check out the "square plate compound" object. There is a corner piece inside the polar array called "corners" called "bl corner", and I made it into a polar array of four around <120,120,0>. The original "bl corner" doesn't appear. Only the other three. I can go into the array and hide/show the original "bl corner", but it doesn't show up in the STEP or STL file when I export it.

I've never before had an array be missing one of the array elements, so I think we have a regression here.

[greek keys.FCStd.zip](https://github.com/user-attachments/files/16822702/greek.keys.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-31:
One of the elements in the array has been made invisible. Most likely this has happened through user actions.

To correct his:
1. Right-click the array in the Tree view.
2. Select "Toggle array elements".
3. Make the invisible element visible.
4. Repeat steps 1 and 2.

theosib, 2024-08-31:
The array is not expanded. And at no point did I toggle the array elements before. Otherwise I'd have noticed and figured out something about toggling it again.

Toggling the array elements did reveal that one was hidden. But this doesn't explain how one of them got hidden in the first place.

Roy-043, 2024-08-31:
Upon further testing I see now that you can select an element in an unexpanded array and make that element invisible. This is what you have experienced and what has caused your confusion.

So a possible solution would be to disallow this for unexpanded arrays.

FEA-eng, 2026-04-24:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260421 (Git shallow)
Build date: 2026/04/21 23:21:45
Build type: Release
Branch: (HEAD detached at 449725202)
Hash: 449725202d0ab581516aacad59d62cdce80db82a
Python 3.11.15, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```