FreeCAD/FreeCAD #16167: Can't drag more objects into a Part:Compound

theosib, 2024-08-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I have a Union in Part, I can drag more objects into it.

But this doesn't work for Compound. Dragging something into the compound has no effect.

This seems like an inconsistency.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-31:
A Part Compound is a low level container. You can only drop objects with a shape. You cannot drop Std_Groups or Std_Parts for that reason.

luzpaz, 2024-08-31:
Is this a FR then ? or by design and the UI should let user know this is by design ?

theosib, 2024-08-31:
I was trying to add a simple extruded sketch, but dragging it to the compound failed silently. There wasn't even a console message.

Incidentally, there ought to be some means to remove an element from a container. Deleting it is fine if it's the only instance, but if an object is used multiple places, we don't want to actually delete it.

Roy-043, 2024-08-31:
That is indeed a special case. You can drop an individual sketch, you can drop the Part Extrude, but not both.

Roy-043, 2024-08-31:
> Incidentally, there ought to be some means to remove an element from a container.

You can drag objects from containers and drop them on the document node.

theosib, 2024-09-01:
> > Incidentally, there ought to be some means to remove an element from a container.
> 
> You can drag objects from containers and drop them on the document node.

I actually tried that, and it didn't work. For example, I tried to drag the object below the compound or between the compound and the object above it. None of these things worked.

theosib, 2024-09-01:
> That is indeed a special case. You can drop an individual sketch, you can drop the Part Extrude, but not both.

I actually just tried to drag the extrude. The sketch is just linked in the extrude.

Roy-043, 2024-09-01:
> None of these things worked.

The document node is a single entry in the Tree view. It is the entry with this icon: https://wiki.freecad.org/images/d/df/Document.svg

theosib, 2024-09-01:
Yeah, I was trying to drag it between two other objects at the top level. But it'll only give me one of those "I'll insert here" lines if I'm all the way at the top. I could see this being a problem if there are too many objects in the main document so that the document object is not on screen.

FEA-eng, 2026-10-25:
See [this](https://github.com/FreeCAD/FreeCAD/issues/16167#issuecomment-3401994429)