FreeCAD/FreeCAD #16170: Double click .STEP file from File Explorer and it opens the Uninstall Dialog

Forresto0, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I double click on a .STEP file or .IGS file using the File Explorer, it opens the Uninstall Dialog for FreeCAD (see picture). If I use the Open files dialog in FreeCAD, the .STEP file opens correctly. This is my first Issue: Report  Problem ever. Therefore I did not search to see if similar issue already exists. Github's search function confused me. I'll do better next time.
[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.15
  * QuickMeasure 2022.10.28
  * SearchBar 1.0.2
[/code]
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-31:
Thanks for posting. Is this some sort of MIME type issue that links those formats to the uninstaller instead of the main FreeCAD app ? 

Can you also test 0.22.x dev version ?  (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Syres916, 2024-08-31:
@luzpaz the dev builds aren't installers so they won't have any effect on a Windows box. I'm guessing there was something in the NSIS config of the 0.21.x installer but I've not been able to test it yet.

Syres916, 2024-08-31:
On Windows 7 there is no association set by the FreeCAD Installer, if I double-click IGS, IGES or STEP files, Windows states nothing is associated with this file type, do you want to specify an application to open it with?

My STP files are associated with Notepad because I specifically asked for it to be the case in the past as I was investigating the contents of multiple files.

It is my understanding that [this](https://github.com/donovaly/FreeCADInstProj/blob/master/setup/configure.nsh) is the file that NSIS uses for file association

Forresto0, 2024-09-02:
I'm not sure if you were asking me about the MIME thingy. But I figured out
how to get the 0.22.x dev version, I think.
This issue *persists*. See pic below.

*Version:*
FreeCAD_weekly-builds-38555-conda-Windows-x86_64-py311

[image: Window_2024-09-02 18_40_17_Uninstall_Prompt.png]

luzpaz, 2024-09-02:
[image]

@Forresto0 better to respond directly on github then via email (it messes with markdown formatting)

Forresto0, 2024-09-05:
@luzpaz I have a correction. 
Using the dev version, .STEP and .IGS files open from file explorer correctly, no issues. My bad. Operator error.

*Version:*
FreeCAD_weekly-builds-38555-conda-Windows-x86_64-py311

luzpaz, 2024-09-05:
Closing ticket

P.S. What was the operator error ?

closed 2024-09-05 by luzpaz