FreeCAD/FreeCAD #16172: Vertical dimension text has wrong orientation or side

pure-machinery, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I get varied results when I try to add a couple of vertical dimensions on this shaft.  They should all be facing the same side as per the picture.

A picture i found online: 
[image]

Here's a video of it:

[Screencast from 2024-08-31 12-17-15.webm](https://github.com/user-attachments/assets/37057c35-1202-45cc-b3b9-dc652afd632f)

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git) AppImage
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BillOfMaterials 0.1.4.2
  * sheetmetal 0.4.24
  * fasteners 0.5.27
  * A2plus 0.4.66
  * fcgear 1.0.0
  * OpenTheme 2024.8.17
  * IconThemes
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

benj5378, 2024-08-31:
@pure-machinery send file

WandererFan, 2024-08-31:
I can not duplicate this.   Is this repeatable?  Need step by step instructions.
[image]

pierreporte, 2024-08-31:
I can reproduce with the following steps, assuming two horizontal edges:

1. Start `TechDraw_Dimension`
2. Click on the lower edge then on the upper edge → text is on the left side of the dimension
3. Start `TechDraw_Dimension` again
4. Click on the upper edge, then on the lower edge → text is on the right side of the dimension

Syres916, 2024-08-31:
@pure-machinery what I found that's not exactly clear in the report is this only applies to using the Single Tool for dimensioning. If I use Separate Tools the _DistanceY_ type is always used for `TechDraw_VerticalDimension` and therefore the text is orientated correctly every time whereas the SingleTool `TechDraw_CompDimensionTools` creates a _Distance_ type which can result in the text incorrectly orientated depending on selection order.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38561 (Git)
Build type: Release
Branch: main
Hash: f88f42a3b94e7303df53e1fe6273797a212549a1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66
```

maxwxyz, 2024-08-31:
I've noticed this in the past as well and can confirm too, with above steps

WandererFan, 2024-08-31:
Does this only happen for vertical dimensions?  I can follow the directions and produce the fail if the dim line runs up and down, but not if it runs left and right.

[dimTextPosition.FCStd.renameMe.zip](https://github.com/user-attachments/files/16825536/dimTextPosition.FCStd.renameMe.zip)

maxwxyz, 2024-08-31:
@WandererFan yes can only reproduce on vertical too

pierreporte, 2024-08-31:
Same, the problem is only for vertical dimensions.

wibenjamin, 2025-05-05:
Hello, the problem still exists for me, see pictures attached. I somehow was able to fix the orientation for some dimensions, but there are still a few that get displayed on the wrong side no matter what I tried.

The build I use is the following:

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

[image]
[image]

pierreporte, 2025-05-05:
@wibenjamin could you test on the latest weekly version?

wibenjamin, 2025-05-06:
> [@wibenjamin](https://github.com/wibenjamin) could you test on the latest weekly version?

Tried on the latest weekly build. Error persists in addition to new not related issues (unable to regenerate section and detail views)

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.41365 (Git)
Build date: 2025/04/22 11:42:50
Build type: Release
Branch: main
Hash: 84b57d67b50ac64be2246418fb4b49406904f3a9
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/108.888/1

WandererFan, 2025-11-28:
Unable to duplicate this in 1.2.0dev.  Selection order has no effect that I can see.  

[image]

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.2.0dev.44449 +3 (Git)
Build date: 2025/11/28 00:29:41
Build type: Unknown
Branch: SheetCyclicDepend
Hash: a1e9998ae8f99294e836bb949dcca4f8cb3ed071
Python 3.10.12, Qt 6.9.3, Coin 4.0.0, Vtk 7.1.1, boost 1_74, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.2.3, OCC 7.8.0
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * A2plus 0.4.68
  * sheetmetal 0.7.58
  * fasteners 0.5.43

```

WandererFan, 2025-11-28:
Can not duplicate on this version either:

```
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0rc1.20251125 (Git shallow) AppImage
Build date: 2025/11/25 17:53:08
Build type: Release
Branch: (HEAD detached at 5376760)
Hash: 53767601ea65de7cd9f6590e7d5856afc0841635
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Canada (en_CA)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/118.229/1.19792
Installed mods: 
  * A2plus 0.4.68
  * sheetmetal 0.7.58
  * fasteners 0.5.43
```

luzpaz, 2025-11-28:
@pure-machinery 👆  can you still reproduce on 1.1RC1?  https://github.com/FreeCAD/FreeCAD/releases/tag/1.1rc1

maxwxyz, 2025-11-29:
I'm not able to reproduce either.

commit 93e685bd (2024-09-02): [TD]fix dim text wrong side of dim line (fix #16172)
closed 2025-11-29 by maxwxyz