FreeCAD/FreeCAD #16174: PartDesign Pad: Reverse option when using Up to Face type is enabled

leoheck, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Does it make sense?

If it does, then the feature is not working.
If not, then the "Reverse" option should be removed.

See the `Pad001` in this example: [test1.zip](https://github.com/user-attachments/files/16825018/test1.zip)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git) AppImage
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Anything else?

This isse https://github.com/FreeCAD/FreeCAD/issues/15461 was split in 3.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-31:
IMO same as in https://github.com/FreeCAD/FreeCAD/issues/16175 I don't see a usecase. Ideally the checkbox should be inactive.
@pierreporte FYI

pierreporte, 2024-08-31:
@maxwxyz Yes, the idea is to disable the checkbox because it is misleading.