FreeCAD/FreeCAD #16175: PartDesign Pad: There is no Symmetric option when using Up to Face type

leoheck, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Does it make sense?

For me it does.. it pads to a face, and then it creates the same pad to the other side, like a mirror. 

See, there is no Symetric option when using Up-To-Face:
[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git) AppImage
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

This isse https://github.com/FreeCAD/FreeCAD/issues/15461 was split in 3.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-31:
What should be the result, could you share a demonstration? When the pad is intersecting the selected face, should this be mirrored or the distance? I cannot make up a use case or do you want to pick two opposite faces?
@pierreporte any input?

pierreporte, 2024-08-31:
I also have no idea of a use case. The only thing would be with up to first, that when marked as symmetrical, would go up to first on both sides, that can be at different distances and have different shapes.

leoheck, 2024-08-31:
@maxwxyz, when you Pad something, what is the use case? You pad to add some mass, right? When you check th Reverse, what does it do? Grow to the other side. And when you mark it Symmetrical, what do you expect? The mass growing on both sides the same length/distance.

FlachyJoe, 2024-09-01:
IMHO it's not needed because you can still use a PartDesign_Mirrored.

leoheck, 2024-09-01:
The same for the other pad types, then.

herrdeh, 2024-09-06:
I'm in favour of this feature. My use case is an enforcement in a rounded body which needs to go symmetrical to both sides. Here is one side - that's easy:
[image]
But to produce the other half turned out to quite be a nightmare. I could not use the same sketch again, neither facebinders, only a copy of the 1st sketch worked.

When doing it symmetric, there need to be two "select face" boxes.

Cheers,
Wolf

leoheck, 2024-09-06:
> When doing it symmetric, there need to be two "select face" boxes.

If you have to select 2 faces then it is not symmetrical.

herrdeh, 2024-09-07:
True.
So something like "two-sided" or "both-sided" might be a better wording.

pierreporte, 2024-09-07:
@herrdeh Two sided extrusions are common in CAD programs and would be legitimate. The user could select up to face (and the face) for each side, or different modes.