FreeCAD/FreeCAD #16177: Part Design Pipe: Adding edges to the Additive/Subtraction Pipe is a multi-click operation.

leoheck, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I could not find the issue discussing the improvement of the method to select edges of some Features.
So I am not sure this will be revamped to 1.0. If not, then at least the Add Edge operation should be improved a bit.

Right now, for each edge I  have to add, I have to click the Add Edge button again. 

The operation is this:
1. Move the cursor to the Add Edge button, click on it.
2. Move the cursor to the Edge, click on it.

If I have 2 edges this is already a tedious work. Imagine if we have 10. 

Note: The Remove Edge operation was already improved since the Delete key is working again, and also the focus remains in the list, so I can hit Dell, and move the selection easily with the keyboard.

Here is the playground to evaluate this.
Check it on the Body named Clone on the AdditivePipe feature. 
[test2.zip](https://github.com/user-attachments/files/16825085/test2.zip)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git) AppImage
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-02:
@leoheck Can you edit the OT and add your suggestion how the UI should be optimized ?

leoheck, 2024-09-03:
@luzpaz , my suggestion is to use something similar of what we have on Chamfer/Fillet tools. On those features we can select multiple edges, and also deselect/remove items easily.