FreeCAD/FreeCAD #16178: Sketcher: Now randomly favoring "point on object" over "coincident" sometimes when auto-constraining

theosib, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the dev builds, I have encountered WAY more "wire not closed" errors than in previous versions. (FreeCAD is still really bad at helping the user find those, but that's a separate problem.) What I'm observing is that on those cases, the Sketcher auto-created a point-on-object constraint when I had unambiguously started a new line on an endpoint.

Before posting this bug report, I asked about it in Discord, and at least one other user has observed a similar phenomenon.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-31:
PLease add an example file or steps how to reproduce. Which tools are you using that create malformed constraints? Does it appear when editing the lines (trim,..).

theosib, 2024-08-31:
Ok, here's one way I got it to happen. 

- Draw two construction lines that cross each other. 
- Create a real line (A) that starts on one construction line and stops at the intersection of the two construction lines.
- Create another real line (B) that starts on the other construction line and try to drop the end point at that same intersection.

I expected the second end point of A to snap to the end point of B, but instead, it snapped to the construction line.

maxwxyz, 2024-08-31:
Is this the same behavior in 0.21.2 stable?

Roy-043, 2024-08-31:
I can't reproduce. Is this in a specific sketch perhaps? Have you tried renaming your CFG file?

There have not been similar reports on the forum AFAICT.

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

> stops at the intersection of the two construction lines.

Note that there is no intersection snap AFAIK.

theosib, 2024-09-01:
There is no snap to an intersection. I just moved the end point there. Then when I added the second line and tried to snap its end point to the first line's end point, it attached to the construction line instead.

It was in fact like the 8th test I tried. This is about right as far as odds go. I can make a sketch with like 100 coincident points in it, and it'll look perfect and be 100% constrained, but I can't extrude due to the wire not being closed. So I have to go in and mess around with disabling constraints until I find two points that are perfectly on top of each other due to constraints to other things but not each other.

Other times I'll find that I have points perfectly on top of each other, but there will be degrees of freedom. And sometimes the sketcher will actually show something useful to find them, and it'll turn out that one line's end point will have gotten constrained to a line that goes through the point it's supposed to be attached to, but it's attached to the line instead.

Other times I'll have points perfectly on top of each other, and it'll turn out that a combo of constraints will have two points exactly on top of each other but it's a combo of something like vertical and horizontal constraints. So although the points can't move away from each other and they're perfectly coincident, there isn't an explicit coincident constraint, so the wire "isn't closed."

Basically, this "wire not closed" thing is a plague that lots of people complain about, FreeCAD doesn't do a good job of showing where the openings are, sometimes there aren't really openings, and it's being really picky, etc. And the one I'm talking about is just one of many reasons why this goes wrong.

Should I file a bug report on Sketcher failing to identify indirect wire closures?

krushia, 2024-09-01:
I experienced a glitch like this a couple weeks ago, but it fixed itself and I never filed a report. I suspected the mouse coordinates of the point (or maybe just the auto-constraint?) weren't being picked until _after_ the mouse was clicked and released. Below is my unfinished report found in clipboard history:

> Not sure what's happening, but sometimes when drawing a new line with one endpoint on an existing point (expected to create an automatic coincident constraint as evidenced by the X icon showing), the position doesn't register where clicked. This manifests as either:
> 
> 1. A coincident constraint is NOT created, and the endpoint is placed very near to the pre-existing point, in the direction that the mouse is moved after click.
> 2. A coincident constraint IS created, and the joined points are dragged slightly in the direction that the mouse is moved after click.
> 3. Some other constraint is created, as if the click happened somewhere else near the point.
> 
> Notes
> 
> - This is most apparent with a computer under heavy CPU load.
> - It does not occur when adding lines to a sketch that is already fully constrained.
> - Snap to grid isn't on.

julia-m-stephenson, 2024-09-16:
I have seen this issue with RC1 when doing something as simple as creating a single slot in a sketch and getting wire not closed when trying to make a slot shaped hole.

A slot is two centered curves with tangential lines connected the two curves. 

I found that the although tangential constraints exist the coincident constraints do not. This is very frustrating and seems happen every time you create a curve tangent to a line. 

I will add a simple example I think this has been an issue for a while I've come across it without realising thecause in daily builds.

maxwxyz, 2024-09-16:
@PaddleStroke FYI

julia-m-stephenson, 2024-09-17:
Just another data point I tried recreating the slot example above from scratch with a freshly opened RC1 instance and it worked no problem I tried again and I got the wire not is not closed. 

So it doesn't seem to happen each time you create similar geometry :-( obviously a pain to debug.

Here are two more projects. SlotExample2 shows the failure and SlotExample3 works fine. Hope this is of some use.
[Slotexample23.zip](https://github.com/user-attachments/files/17022625/Slotexample23.zip)
[image]
[image]

julia-m-stephenson, 2024-09-18:
I seem to have a lot of problems with shapes that include tangents having wire not closed e.g I was trying to create an end cap for this shape note two sets of tangents.

[image]

If I used external geometry to trace the shape I couldn't ever get a closed wire. In the end I drew the shape (three lines and two curves) and then added the external geometry and manual made the joint vertices and curve centre vertices on my free hand shape match the external geometry.

[image]

This seems to break the normal way of using external geometry and is very frustrating. I have tried deleting my config and starting afresh but that hasn't changed anything. 

I'm surprised more people aren't seeing this issue is it really frustrating.

Roy-043, 2024-09-19:
After checking SlotExample2 and SlotExample3, and lots of tests, I think have found a reproducible scenario for that problem.

Steps:
1. Switch the grid and grid snap on.
2. Pick a grid point for the 1st point of the slot.
3. Pick another grid point, to the right of the previous point, for the 2nd point.
4. Pick a 3rd point to define the width of the slot.

The 1st and 2nd point must have the same Y coordinate. If you do not snap to the grid this will rarely occur, which explains why it seems so random.