FreeCAD/FreeCAD #16180: CAM New CAM Simulator Tool Size/Shape incorrect

firebladed, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Running The New Simulator with a Slot Saw generates a tool of incorrect size and shape

1 Load [Freecad CAM Simulator tool generation bug.zip](https://github.com/user-attachments/files/16825268/Freecad.CAM.Simulator.tool.generation.bug.zip)
2. Run New Simulator
3. Start Sim

Simulator Tool
[image]
Actual Tool Size & Shape
[image]

Don't know if this tool is defined with wrong properties, as don't know what supported properties are

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * Assembly4 0.50.15
  * Assembly4.backup1714492159.8689911 (Disabled)
  * boltsfc 2022.11.5
  * btl 0.9.9
  * EM 2.1.1
  * fasteners 0.5.25
  * Help 1.0.3
  * kicadStepUpMod
  * OpticsWorkbench 1.0.18
  * OpticsWorkbench.backup1714492175.9109938 (Disabled)
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-31:
@shaise FYI

shaise, 2024-08-31:
Thanks I will take a look.

MarcusMWilliams, 2024-08-31:
It's Real

shaise, 2024-08-31:
Yes, the reason is that the simulator is based on that the tool is at position 0,0,0 and it is not
if you click the tool in the tree and reset its position, this will be fixed:
[image]

Anyway, I will try to find a way to compensate for that

shaise, 2024-08-31:
OK, I have fixed it so it will handle tools that are not in the reset position. It is added to PR #16118

FEA-eng, 2024-09-02:
@maxwxyz Apparently, this can be closed as fixed by #16118.

closed 2024-09-02 by maxwxyz