FreeCAD/FreeCAD #16181: update onsdelsolver git submodule

looooo, 2024-08-31:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For compiling on windows with clang I need some recent commits of the onsdel-solver git submodule. Is it possible to update the submodule to latest main branch?

### Full version info

```shell
no version info available as compilation fails.
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2024-08-31:
@PaddleStroke can we safely update? To HEAD? Or to a specific revision?

PaddleStroke, 2024-09-01:
I have not tested the latest main of the solver. I'll give it a go

luzpaz, 2024-09-03:
heads-up: #16253

huguesdpdn-aerospace, 2024-09-08:
Hello !

Please be aware that on [Ondsel repository](https://github.com/Ondsel-Development/OndselSolver/commits/main/), a tremendous amount of PRs about compilation, warnings compilation, includes issues, etc. including both UNIX and Windows platforms were recently merged.

Overall, recently, Ondsel merged the following PRs:
- https://github.com/Ondsel-Development/OndselSolver/pull/79
- https://github.com/Ondsel-Development/OndselSolver/pull/77
- https://github.com/Ondsel-Development/OndselSolver/pull/76
- https://github.com/Ondsel-Development/OndselSolver/pull/75
- https://github.com/Ondsel-Development/OndselSolver/pull/74
- https://github.com/Ondsel-Development/OndselSolver/pull/73
- https://github.com/Ondsel-Development/OndselSolver/pull/72
- https://github.com/Ondsel-Development/OndselSolver/pull/70
- https://github.com/Ondsel-Development/OndselSolver/pull/68
- https://github.com/Ondsel-Development/OndselSolver/pull/64
- https://github.com/Ondsel-Development/OndselSolver/pull/59
- https://github.com/Ondsel-Development/OndselSolver/pull/53

Also, some minor fixes has been provided since last April in the OndselSolver repository.

Thanks @chennes for being the author of 2 of them and @PaddleStroke for reviewing some of them.

So @PaddleStroke , if you wish to test the last `OndselSolver` version, especially under Windows, it would be awesome.

If this is good, I am interested to bumb OnsdelSolver version before the V1.0 release since I am extremely cramped with all compilation warnings ; which avoid me to work easely and properly on my latest developments on FreeCAD. Others libraries compilation fixes are also ongoing.

Beside, @chennes + @PaddleStroke : I didn't find on FreeCAD repository how to bump to a newer OndselSolver version. I don't see any hardcoded commit hashes or tags/releases in OndselSolver. Shouldn't it be automated to the latest version with the `git --recurse-submodules` option ? Or do we need to specify somewhere where to update the Ondsel version to use ?

Thank you

chennes, 2024-09-09:
Working with submodules is much like working with manually checked-out git repos: you can always update your local checkout to whatever version you like, even if FreeCAD main does not do so. I do, for exactly the reason you state -- so that I can silence those compilation warnings.

To update the one that everyone uses, you check out the correct revision locally, then in your FreeCAD repo you will see that it is marked as changed by git. Add it, commit it, push it, and create a PR. Generally speaking I wait for @PaddleStroke to do that 😁 .

PaddleStroke, 2024-09-09:
I'm on it now

StefanBruens, 2024-09-10:
This breaks the GCC minimum requirement as documented in  
https://wiki.freecad.org/FreeCAD_1.0_Development_Cycle

> std::filesystem is not well-supported by GCC 7.5 -- continue to use boost::filesystem instead

Note, the filesystem usage is for debug output only:
https://github.com/Ondsel-Development/OndselSolver/blame/91f70382beeb58b32432b5a82da3802e4c19199c/OndselSolver/ASMTAssembly.cpp#L362-L364

chennes, 2024-09-11:
@aiksiongkoh or @PaddleStroke can you please address @StefanBruens's comment and push a new submodule pin?

huguesdpdn-aerospace, 2024-09-11:
Hello,

I don't know if by "blaming" you mean to correct this issue or to rollback but as I am in doubt about your intentions and a bit scared, please do NOT rollback any OndselSolver version since I am struggling hard to debug FreeCAD with the compilation warnings and that I am currently solving all Xerces-C warnings compilation in order to solve #16316  .

Instead, as far as I can see, `#include <filesystem>` only works since GCC >= 8 (see [[Answered] - Why does GCC not seem to have the filesystem standard library?](https://stackoverflow.com/a/45867491/12642493), can I suggest to create a new PR in OndselSolver with the following proposition ?

```
#if __GNUC__ >= 8
    // Use std::filesystem
#else
    // Use boost::filesystem
#endif
```

As far as I can see, we only need to patch in this file:
- [OndselSolver/ASMTAssembly.cpp](https://github.com/Ondsel-Development/OndselSolver/blob/889196e3267597127b5889572b0c86f9316e16f0/OndselSolver/ASMTAssembly.cpp) =>Need to implement my above block code at line 362

And about this file, let's even patch it:
- [OndselSolverMain/OndselSolver.cpp](https://github.com/Ondsel-Development/OndselSolver/blob/889196e3267597127b5889572b0c86f9316e16f0/OndselSolverMain/OndselSolver.cpp) => @PaddleStroke : can you remove the `#include <filesystem>` ? Seems unused (no debug in file)

Would this solution be okay for you @PaddleStroke + @chennes ?

chennes, 2024-09-11:
There is no plan to roll back to a previous revision, @huguesdpdn-aerospace -- I think your proposed solution is exactly the right one.

adrianinsaval, 2024-09-18:
why do we want to support gcc 7.5? what supported system is using that? ubuntu 20.04 is supposed to be our baseline and searching for the package gcc 9.3 seems to be their default version, with gcc 10 also available https://packages.ubuntu.com/search?suite=focal&searchon=names&keywords=gcc

is there any problem still with std::filesystem in gcc > 9.3 ?

adrianinsaval, 2024-09-18:
the wiki page mentioned was actually created for the development cycle of what ended up being 0.21 not 1.0 so I don't think we should take it so seriously, IMO we can reasonably bump minimum requirement to gcc 9.3

huguesdpdn-aerospace, 2024-09-20:
Created [Fix compilation under gcc7.5 #81](https://github.com/Ondsel-Development/OndselSolver/pull/81)

No `boost` library in OndselSolver. So with GCC < 8, the debug output won't be displayed. But I think this is no big deal if it is not display.

maxwxyz, 2024-10-30:
is this still relevant?

StefanBruens, 2024-10-30:
> is this still relevant?

Yes. The PR is incomplete, and has been ignored mostly.

huguesdpdn-aerospace, 2024-11-04:
PR merged in OndelSolver repo :)

See https://github.com/Ondsel-Development/OndselSolver/pull/81

huguesdpdn-aerospace, 2024-11-06:
PR linked to this issue: https://github.com/FreeCAD/FreeCAD/pull/17700

closed 2024-11-22 by chennes