FreeCAD/FreeCAD #16182: [Base] Segmentation fault at startup

benj5378, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

```
#0  __strrchr_evex () at ../sysdeps/x86_64/multiarch/strrchr-evex-base.S:83
#1  0x00007ffff2bfd83e in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#2  0x00007ffff29bf2f3 in SbkType_FromSpecWithBases ()
    at /lib/x86_64-linux-gnu/libshiboken2.cpython-312-x86_64-linux-gnu.so.5.15
#3  0x00007ffff29cfccf in Shiboken::Enum::newTypeWithName(char const*, char const*, _typeobject*) ()
    at /lib/x86_64-linux-gnu/libshiboken2.cpython-312-x86_64-linux-gnu.so.5.15
#4  0x00007ffff29cff96 in Shiboken::Enum::createScopedEnum(SbkObjectType*, char const*, char const*, char const*, _typeobject*) () at /lib/x86_64-linux-gnu/libshiboken2.cpython-312-x86_64-linux-gnu.so.5.15
#5  0x00007fff70451078 in init_QtCoreQt(_object*) ()
    at /home/bensay/.local/lib/python3.12/site-packages/PySide2/QtCore.abi3.so
#6  0x00007fff704cbe3f in PyInit_QtCore ()
    at /home/bensay/.local/lib/python3.12/site-packages/PySide2/QtCore.abi3.so
#7  0x00007ffff2cd641f in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#8  0x00007ffff2cd33aa in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#9  0x00007ffff2bdc765 in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#10 0x00007ffff2b19dec in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#11 0x00007ffff2b82d6e in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#12 0x00007ffff2b82fb1 in PyObject_CallMethodObjArgs () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#13 0x00007ffff2cd4a12 in PyImport_ImportModuleLevelObject () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#14 0x00007ffff2b23a6c in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#15 0x00007ffff2c9d1ff in PyEval_EvalCode () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#16 0x00007ffff2c99180 in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#17 0x00007ffff2bdc4ac in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
--Type <RET> for more, q to quit, c to continue without paging--c
#18 0x00007ffff2b19dec in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#19 0x00007ffff2b82d6e in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#20 0x00007ffff2b82fb1 in PyObject_CallMethodObjArgs () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#21 0x00007ffff2cd4a12 in PyImport_ImportModuleLevelObject () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
[...]
#91 0x00007ffff2b82afc in PyObject_Vectorcall () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#92 0x00007ffff2b1d5ee in _PyEval_EvalFrameDefault () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#93 0x00007ffff2c9d1ff in PyEval_EvalCode () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#94 0x00007ffff2cf7af8 in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#95 0x00007ffff2cf7c1c in  () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#96 0x00007ffff2cfb234 in PyRun_StringFlags () at /lib/x86_64-linux-gnu/libpython3.12.so.1.0
#97 0x00007ffff3790240 in Base::InterpreterSingleton::runString[abi:cxx11](char const*)
    (this=0x55555562c630, sCmd=0x7ffff6eaaba0 <FreeCADGuiInit> "#", '*' <repeats 75 times>, "\n#*   Copyright (c) 2002,2003 Jürgen Riegel <juergen.riegel@web.de>         *\n#*", ' ' <repeats 43 times>...)
    at /home/bensay/Development/FreeCAD/src/Base/Interpreter.cpp:249
#98 0x00007ffff645b590 in Gui::Application::runInitGuiScript() ()
    at /home/bensay/Development/FreeCAD/src/Gui/Application.cpp:1981
#99 0x00007ffff6658fb7 in Gui::StartupPostProcess::showMainWindow() (this=0x7fffffffd7b0)
    at /home/bensay/Development/FreeCAD/src/Gui/StartupProcess.cpp:440
#100 0x00007ffff6657782 in Gui::StartupPostProcess::execute() (this=0x7fffffffd7b0)
    at /home/bensay/Development/FreeCAD/src/Gui/StartupProcess.cpp:231
#101 0x00007ffff645c491 in Gui::Application::runApplication() ()
    at /home/bensay/Development/FreeCAD/src/Gui/Application.cpp:2141
#102 0x000055555557ac86 in main(int, char**) (argc=1, argv=0x7fffffffde38)
    at /home/bensay/Development/FreeCAD/src/Main/MainGui.cpp:300

```

Doesn't the crash look like it's caused by trying to interpret
```
# ***************************************************************************
# *   Copyright (c) 2001,2002 Jürgen Riegel <juergen.riegel@web.de>         *
# *                                                                         
```

### Full version info

```shell
FreeCAD 0.22.0 Revision: 38555 (Git)

Library versions:
boost    1_83
Coin3D   4.0.2
Eigen3   3.4.0
OCC      7.6.3
Qt       5.15.13
Python   3.12.3
PySide   5.15.13
shiboken 5.15.13
VTK      
xerces-c 3.2.4
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
Are you using your own build or a weekly build? What system are you using it on?

benj5378, 2024-09-01:
> Are you using your own build or a weekly build? What system are you using it on?

Personal build on my laptop. Linux, Ubuntu 23.04. Maybe this should be added to the version info (`FreeCAD --verbose --version`)?

bdieterm, 2024-09-01:
Seems like a problem caused while executing src/Gui/FreeCADGuiInit.py.
In this script, all the InitGui.py files from the mods are executed as well.

First, disable all your mods by renaming the `~/.local/share/FreeCAD/Mod` folder and try again to run FreeCAD.

If this does not solve the problem, then try to debug the FreeCADGuiInit.py script as described here (using breakpoints etc.):
https://wiki.freecad.org/Pyzo#Debugging_Start-Up_of_FreeCAD_Modules

luzpaz, 2024-09-09:
> First, disable all your mods by renaming the `~/.local/share/FreeCAD/Mod` folder and try again to run FreeCAD.

Another usecase for #15710

benj5378, 2024-09-29:
This might be related to `import shiboken2` being incompatible with python3.12

benj5378, 2024-09-29:
Seemed like I had two shiboken packages installed.
PyPi package not compatible with Python 3.12
Ubuntu Package compatible with Python 3.12

Removing the PyPi package fixed the crash.

I think the compile didn't give warning or fail, because it used the compatible Ubuntu Package. While when starting FreeCAD, the incompatible PyPi package was used, causing the crash.

No crash, though still errors
```During initialization the error "libshiboken2.abi3.so.5.13: cannot open shared object file: No such file or directory" occurred in /home/bensay/Development/FreeCAD-build/Mod/Measure/InitGui.py```

closed 2024-09-29 by benj5378

benj5378, 2024-09-29:
@chennes would it be sensible to add
```
from shiboken2 import __version__ as t
if t < 5.15.13 and pythonVersion > 3.12:
    throw RuntimeException("Incompatible shiboken version")
```
?

bgbsww, 2024-09-29:
Quibble:   >= 3.12  or > 3.11   maybe?   Or is this a 3.13 error?   Actually the < too.

idkSeth, 2024-10-02:
> There were also some guy on Discord having shiboken2 import issues - however, I think it was at compile time?

If you're mentioning me, I also had issues importing both shiboken2 and thus PySide in a Python console. I followed the installation commands and installed PySide through my package manager, apt.