FreeCAD/FreeCAD #16183: Crash when deleting chamfer

nirvdrum, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've added a chamfer to a body that I'm unable to delete. Whenever I try to delete it, FreeCAD crashes.

I've attached a design file where the last mutation in the body is a chamfer named "Chamfer". Attempting to delete it crashes FreeCAD 0.22-dev to crash on both Linux and macOS. I've included the trace from the segfault on Linux below:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7e9e57633320]
#1  /tmp/.mount_FreeCA1cRoY0/usr/lib/Measure.so(+0x3f516) [0x7e9e4807d516]
#2  0x7e9e4808e10f in Measure::MeasureDistance::getShape(App::PropertyLinkSub*, TopoDS_Shape&) from /tmp/.mount_FreeCA1cRoY0/usr/lib/Measure.so+0x1bf
#3  0x7e9e4808e975 in Measure::MeasureDistance::execute() from /tmp/.mount_FreeCA1cRoY0/usr/lib/Measure.so+0x475
#4  0x7e9e59662492 in App::DocumentObject::recompute() from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADApp.so+0xc2
#5  0x7e9e5963c4fc in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADApp.so+0x32c
#6  0x7e9e59648ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADApp.so+0x625
#7  0x7e9e596c773e in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADApp.so+0x2ee
#8  0x7e9e596c7854 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADApp.so+0x24
#9  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7e9e59e06068]
#10  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7e9e59de5b5c]
#11  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7e9e59df1532]
#12  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7e9e59eabd1d]
#13  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7e9e59eab794]
#14  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7e9e59ecabf3]
#15  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7e9e59ec669a]
#16  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7e9e59eba9c9]
#17  0x7e9e5913d9fc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADBase.so+0x6c
#18  0x7e9e5a789a84 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0x94
#19  0x7e9e5a7aaa60 in StdCmdDelete::activated(int) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0xc20
#20  0x7e9e5a793c08 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0x2a8
#21  0x7e9e5a793fc4 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0x134
#22  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7e9e57aeb6f2]
#23  0x7e9e5896a462 in QAction::triggered(bool) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x32
#24  0x7e9e5896ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0xab
#25  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5(+0x2e12e2) [0x7e9e58ae12e2]
#26  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5(+0x2e8c34) [0x7e9e58ae8c34]
#27  0x7e9e589aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x178
#28  0x7e9e589703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#29  0x7e9e58978262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#30  0x7e9e5a73e3b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0x99
#31  0x7e9e57ab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Core.so.5+0x108
#32  0x7e9e589763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#33  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5(+0x1c72a5) [0x7e9e589c72a5]
#34  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7e9e589ca0c9]
#35  0x7e9e589703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#36  0x7e9e5a73e3b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0x99
#37  0x7e9e57ab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Core.so.5+0x108
#38  0x7e9e581353ec in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Gui.so.5+0x84c
#39  0x7e9e5810b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Gui.so.5+0xac
#40  /tmp/.mount_FreeCA1cRoY0/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x7e9e52754bea]
#41  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x7e9e55f10ad7]
#42  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x7e9e55f13ac7]
#43  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x7e9e55f140ce]
#44  0x7e9e57b0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Core.so.5+0x66
#45  0x7e9e57ab570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Core.so.5+0x11b
#46  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5(+0x2e675f) [0x7e9e58ae675f]
#47  0x7e9e58ae684f in QMenu::exec(QPoint const&, QAction*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x2f
#48  0x7e9e5a9aba15 in Gui::TreeWidget::contextMenuEvent(QContextMenuEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0xab5
#49  0x7e9e589aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x178
#50  0x7e9e58a56c6e in QFrame::event(QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x1e
#51  0x7e9e57ab68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Core.so.5+0x7b
#52  0x7e9e5897039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#53  0x7e9e58977200 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5+0x340
#54  0x7e9e5a73e3b9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libFreeCADGui.so+0x99
#55  0x7e9e57ab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Core.so.5+0x108
#56  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5(+0x1c78c9) [0x7e9e589c78c9]
#57  /tmp/.mount_FreeCA1cRoY0/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7e9e589ca0c9]
[... 29 lines cut ...]
Locale: English/United States (en_US)

OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Anything else?

This might be related to #12645.

Attached: [chamfer-crash-design-file.tar.gz](https://github.com/user-attachments/files/16827143/chamfer-crash-design-file.tar.gz)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
@bgbsww FYI

benj5378, 2024-09-01:
Can confirm.

```cpp
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff67aa69f in Base::BaseClass::isDerivedFrom<App::Link> (this=0x0)
    at /home/bensay/Development/FreeCAD/src/Base/BaseClass.h:167
167	        return getTypeId().isDerivedFrom(T::getClassTypeId());
(gdb) bt
#0  0x00007ffff67aa69f in Base::BaseClass::isDerivedFrom<App::Link>() const
    (this=0x0) at /home/bensay/Development/FreeCAD/src/Base/BaseClass.h:167
#1  0x00007fffcc1851c7 in Measure::MeasureBaseExtendable<Part::MeasureDistanceInfo>::getMeasureInfo(App::SubObjectT&) (subObjT=...)
    at /home/bensay/Development/FreeCAD/src/Mod/Measure/App/MeasureBase.h:119
#2  0x00007fffcc1c1e2a in Measure::MeasureDistance::getShape(App::PropertyLinkSub*, TopoDS_Shape&) (this=0x555559742390, prop=0x555559742930, rShape=...)
    at /home/bensay/Development/FreeCAD/src/Mod/Measure/App/MeasureDistance.cpp:170
#3  0x00007fffcc1c216c in Measure::MeasureDistance::execute()
    (this=0x555559742390)
    at /home/bensay/Development/FreeCAD/src/Mod/Measure/App/MeasureDistance.cpp:201
#4  0x00007ffff4b04a0c in App::DocumentObject::recompute()
    (this=0x555559742390)
    at /home/bensay/Development/FreeCAD/src/App/DocumentObject.cpp:162
#5  0x00007ffff4a0fc33 in App::Document::_recomputeFeature(App::DocumentObject*) (this=0x5555580b9130, Feat=0x555559742390)
    at /home/bensay/Development/FreeCAD/src/App/Document.cpp:3141
#6  0x00007ffff4a0d402 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int)
    (this=0x5555580b9130, objs=std::vector of length 0, capacity 0, force=false, hasError=0x0, options=0)
--Type <RET> for more, q to quit, c to continue without paging--c
    at /home/bensay/Development/FreeCAD/src/App/Document.cpp:2868
#7  0x00007ffff4bd4f61 in App::DocumentPy::recompute(_object*) (this=0x5555580e0400, args=0x7ffff3546c78 <_PyRuntime+58904>)
    at /home/bensay/Development/FreeCAD/src/App/DocumentPyImp.cpp:578
#8  0x00007ffff4bcc13f in App::DocumentPy::staticCallback_recompute(_object*, _object*) (self=0x5555580e0408, args=0x7ffff3546c78 <_PyRuntime+58904>)
    at /home/bensay/Development/FreeCAD-build/src/App/DocumentPy.cpp:1958
```

benj5378, 2024-09-01:
The issue is on MeasureBase.h:118 where `subObjT.getSubObject()` returns `nullptr`, thus causing a segfault at the later `sub->isDerivedFrom<App::Link>()`

```cpp
Thread 1 "FreeCAD" hit Breakpoint 2.3, Measure::MeasureBaseExtendable<Part::MeasureDistanceInfo>::getMeasureInfo (subObjT=...)
    at /home/bensay/Development/FreeCAD/src/Mod/Measure/App/MeasureBase.h:119
119	        if (sub->isDerivedFrom<App::Link>()) {
(gdb) l
114	    static Part::MeasureInfoPtr getMeasureInfo(App::SubObjectT& subObjT)
115	    {
116	
117	        // Resolve App::Link
118	        App::DocumentObject* sub = subObjT.getSubObject();
119	        if (sub->isDerivedFrom<App::Link>()) {
120	            auto link = static_cast<App::Link*>(sub);
121	            sub = link->getLinkedObject(true);
122	        }
123	
(gdb) p subObjT
$10 = (App::SubObjectT &) @0x7fffffffaa00: {<App::DocumentObjectT> = {document = "Doorbell_Hole_Cover", object = "Body", label = "Body", property = ""}, 
  subname = "Chamfer.;#181:25;:U2;CHF;:H12ad:8,E.Edge14"}
(gdb) p sub
$11 = (App::DocumentObject *) 0x0
```

benj5378, 2024-09-01:
And this is because
https://github.com/FreeCAD/FreeCAD/blob/2a9cf9e8e6e62086948eb15e24d1e69a54f53d97/src/Mod/PartDesign/App/Body.cpp#L520
return nullptr, because the `PropertyLink` `;#181:25;:U2;CHF;:H12ad:8,E` (`Chamfer.;#181:25;:U2;CHF;:H12ad:8,E.Edge14`) could not be found.

And this is because Edge14 was created by the chamfer. So deleting the chamfer deletes the Edge14, the reference for the measurement.

So this is how it should be fixed. If measurement on recompute can't get the reference (on MeasureBase.h:118), then the recompute should return, and the object should show a broken link error in the GUI. Instead of keeping going with an object that doesn't exist, `nullptr`.

This is really a Measure mod issue. @hlorus

benj5378, 2024-09-01:
I would make this high priority, as the crash will happen whenever a measurement is linked to a deleted geometry.

maxwxyz, 2024-09-01:
I'll label it as blocker then.

nirvdrum, 2024-09-01:
Thanks for the quick investigation. Please feel free to retitle the issue as you see fit. I had seen the measurement reference in the trace, but wasn't sure if that was an internal computation thing or something else.

I can confirm that if I delete the Measurement reference I can then delete the chamfer. Apologies for not catching and trying that earlier.

benj5378, 2024-09-01:
> Apologies for not catching and trying that earlier.

No worries, it wasn't that clear on the traceback I used 2 hours in GDB before realizing myself ;)

bgbsww, 2024-09-01:
I appreciate the work here.  Good to have as many triage & debug people as we can.

hlorus, 2024-09-02:
Thanks alot for the investigation!

commit 62a44a94 (2024-09-02): Measure: Fix crash when removing referenced element

Also improve error messages
Fixes #16183
closed 2024-09-02 by WandererFan