FreeCAD/FreeCAD #16184: Will the Sketch section feature come?

sedatkrn, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

"A development proposal for being able to take reference lines and points from any drawing section taken from a 3D created object."

### Full version info

```shell
After creating an object, we take sketch planes from different regions to perform certain operations and add new shapes. However, we are unable to add the reference geometries of some intersection regions to our sketches on these planes. On this topic, 'Zheng Lei (realthunder)' has developed very useful features in his own FreeCAD branch, and adding these features to the main FreeCAD would be very beneficial.

For more details, please check the forum.
https://forum.freecad.org/viewtopic.php?p=776096#p776096
```

### Subproject(s) affected?

PartDesign

### Anything else?

https://forum.freecad.org/download/file.php?id=272607
https://forum.freecad.org/download/file.php?id=272606

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
@sedatkrn PLease use a concise title and description, the problem is not clear

maxwxyz, 2024-09-01:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/12670

closed 2024-09-01 by maxwxyz