FreeCAD/FreeCAD #16187: No option to save / store multiple user-specific clipping planes in the document

maxwxyz, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When solving
- https://github.com/FreeCAD/FreeCAD/issues/11169
- https://github.com/FreeCAD/FreeCAD/issues/11173

It should be possible to save custom clipping planes in the document, and activate them (similar how measurement objects are saved in the tree).
Double clicking them should allow to change the properties of the clipping planes and update and save them.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
@pierreporte FYI, IMO also a much needed feature for industry work. Currently you have to manually set up the clipping plane each time you activate it.

pierreporte, 2024-09-01:
Yes, this is definitely needed. Is this also about allowing to have multiple active section views at once?

maxwxyz, 2024-09-01:
@pierreporte yes but this is already possible today

pierreporte, 2024-09-01:
It will be necessary to take care that unsaved section views don’t mark the document as modified. Measurement has a similar issue: https://github.com/FreeCAD/FreeCAD/issues/13735.

maxwxyz, 2025-07-20:
In RT branch it is possible to save custom views from clipping planes:

[image]