FreeCAD/FreeCAD #16188: Sketcher: Polyline can't start with other than straight line

Panda631, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It's not possible to start a polyline in the sketcher with e.g. an arc. 
You need at least one straight line to use the M key.

The same with v0.21.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.27
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.8.4
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-09-01:
This limitation makes sense if you consider that for an arc segment a tangent vector is required. You can create an separate arc first and then click on one of its endpoints to continue with a polyline.

PaddleStroke, 2025-03-13:
Potentially at the beginning it could offer a arc mode where you draw an arc same as with the arc tool, but then keeps going into the classic polyline tool.

PaddleStroke, 2025-03-13:
Or another idea is that it start like a 3 point arc.

Triffen, 2025-12-02:
It is still relevant and annoys me greatly since 0.19.

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Gesture/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```