FreeCAD/FreeCAD #16189: VarSet: must enter Name of property first

Roy-043, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding a property to a VarSet you must enter the Name first.

If you don't and click on the down arrow of the Group or the Type combo there are 2 popups.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kriswilk, 2024-10-08:
In addition to the pop-ups, I see no practical reason why the "Value" field ever needs to be hidden.

It should be possible to type/select any field in any order. IMHO, validation should either occur upon clicking "OK" or should be non-invasive, such as how Spreadsheet alias naming works (text turns red for invalid names).

pieterhijma, 2024-10-08:
> In addition to the pop-ups, I see no practical reason why the "Value" field ever needs to be hidden. It should always be visible.
> 
> It should be possible to type/select any field in any order. IMHO, validation should either occur upon clicking "OK" or should be non-invasive, such as how Spreadsheet alias naming works (text turns red for invalid names).

This has to do with the implementation and design choices: The dialog tries to give users the rich editor that belongs to the type. It is only possible to create such an editor when the name, group, and type is known. Effectively, you have to first create a property before you can show the editor. The group is not critical as it can be changed later. If it is possible to rename properties (without deleting them) or perhaps change the type, then the dialog can become more in the sense you describe.

An alternative is reworking the whole editor system but this is a major undertaking.

kriswilk, 2024-10-08:
From an outside perspective, that's bizarre. If it's really impossible to create a rich editor based on a specified type without actually creating a property in the object tree, that in itself is an issue.

I did notice that the property shows up preemptively in the Data view as soon as you enter a name, and wondered why.

This causes other undesirable peculiarities. For instance, type a name such as "myvar" and give it a value. Then return to the name field and backspace, making it "myva". The Value field disappears, and, if you tab/click away to make it reappear, the previous value is lost. To end users, none of this will make sense.

https://github.com/user-attachments/assets/b731c2be-0ddf-48be-b3da-7a2779099a29

pieterhijma, 2024-10-08:
Yes, I know. I wanted the following:
- define a variable *and* provide a value in one flow,
- use the rich, native editors that support expressions as well.

Given the existing code, this is what is possible. I would like to see this changed as well but as I said, that is a major rewrite of the editor code. I think it is already a large improvement given the previous workflow documented in [the description of this issue](https://github.com/FreeCAD/FreeCAD/issues/13038#issue-2194688874) (note that some of it has already changed, for example having to click "Show all"). 

Some other remarks:
- As an example for a rich editor, PropertyColor gives you the native color editor.
- The Python console shows the creation, deletion of properties
- Having the VarSet in the property view while adding variables also gives you an indication what is happening under the hood.

Thanks for bringing it up. If enough users find it problematic, perhaps we get enough support to do the major code change to create editors based on type without having to create an actual property. Then again, the editor will change based on the type, so what if you change the type of color to float? What value should be remembered in that case?

kriswilk, 2024-10-08:
Oh, yes, this is all welcome functionality. I'm sure there will be plenty of helpful feedback once 1.0 is released.

With regard to your specific example, I would NOT expect a value to be preserved if the "Type" changes. However, just changing the name or group of the parameter shouldn't (in principle) affect the value entered.

furgo16, 2024-10-08:
I'm also supportive of VarSets, so please do not take this as criticism to the concept or features. That said, I do feel that the UI is currently suboptimal, even if it is an MVP. Hence this I hope constructive feedback as a user.

> It should be possible to type/select any field in any order. IMHO, validation should either occur upon clicking "OK" or should be non-invasive, such as how Spreadsheet alias naming works (text turns red for invalid names).

I fully agree, and I was thinking the same today. The popups are highly disruptive. If there is a less disruptive and more common pattern available, it would be worth considering it.

My concern is that this should be a simple dialog and a quick operation: if I'm to use varsets to create, say 30 properties, it should be streamlined and effective. My perception is that there is a risk of overengineering it.

I wonder if this might be an opportunity for the Design Working Group to provide feedback (@maxwxyz @pierreporte @obelisk79 and others are in there, I believe)?

In any case, thanks for the work that you do!

kriswilk, 2024-10-08:
> My concern is that this should be a simple dialog and a quick operation: if I'm to use varsets to create, say 30 properties, it should be streamlined and effective. My perception is that there is a risk of overengineering it.

The following is also to be taken pragmatically, not as criticism:

Personally (and I said this in another issue somewhere), until a smoother-working dialog can be adopted for ALL properties (not just VarSets), it might be safer to revert to the standard "Add Property" dialog for now. i.e. this is the one you get by right-clicking the Data view and choosing "Add Property".

If that were to appear when you double-click a VarSet, it would still allow rapid creation of multiple properties. Then you can easily add values in the Data panel. This workflow is already extremely useful.

I know it wouldn't be the fully-realized vision, but something that works with no weirdness for 1.0 might be in the best interest of everyone involved. I worry that new users will get a bad taste from what is really a great feature.

pieterhijma, 2024-10-09:
Oh no, I completely agree with most of the feedback. Feel free to continue giving constructive feedback. I'm happy people are taking a closer look at it! Sometimes with new features though, you're limited with what the code already provides. Large changes are difficult to merge especially if there is no clear support from the community.

pieterhijma, 2025-04-15:
Just a note for myself. I have been working on this and I thought I had a good solution that makes the logic much easier and allows one to enter values immediately, not just after a property has been created. However, it turns out that this has problems because when the property doesn't exist yet, the property editor doesn't know what unit to use, something that was just fixed in #15557. Properties, property editors, and property items are very much entangled... 

A path forward may be to create a temporary property and as soon as a valid name is available rename the property. This creates a dependency on #16222.

Another path forward is to try to find a way to somehow set the value field with the default value and unit without relying on any property.

pierreporte, 2025-04-16:
Agree with @kriswilk and @furgo16.

@pieterhijma If you make a temporary property, please make it in such way that the document isn't marked as modified until it has been fully created with its name. It's always a pain to have “modified” document just because temporary things are done (like measures or opening a sketch without doing anything).

pieterhijma, 2025-04-16:
> [@pieterhijma](https://github.com/pieterhijma) If you make a temporary property, please make it in such way that the document isn't marked as modified until it has been fully created with its name. [...]

Thanks, that's a good point. I'm getting closer to an implementation and the current one does satisfy this.

pieterhijma, 2025-04-16:
I think I have a satisfying solution in #20824. I use a temporary property not registered with the document, so the document is not touched, no rename is necessary and the unit is taken into account as well. The dialog is now much easier to use and although it is still possible to get a popup, it is really difficult to get there (I haven't managed to trigger it).

PR #20824 (2025-04-23): Gui: Fix value field for DlgAddPropertyVarSet
The value field would only appear if a name was provided resulting in a strange user experience.  This commit fixes this and simplifies the logic of the dialog. It is now possible to assign a value first and then come up with a name. 

This PR is a major rewrite of the dialog but it is now much simpler and robust. It is still necessary to create a property, but I now create a temporary property that I can delete after the PropertyItem has been created and the data of the editor has been set.

Closes #16189
Closes #17474
closed 2025-04-23 by kadet1090