FreeCAD/FreeCAD #16190: Clipping plane always cuts the entire document - no option to define only a subset of objects to cut

maxwxyz, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The clipping plane in the 3D view currently always cuts through all present parts, bodies and features.
There is no way to define which objects to cut exclusively or which to exclude from the cut.
The persistent section cut feature allows this but creates new geometry and is very slow to compute on a larger document.

By default it should cut through the entire document (all objects). 
It should be possible to define per section plane
- objects to exclude from being cut
- exclusive objects to be cut
- define if children of the selection are included for the first two options

The selection should be saved when the clipping planes can be stored inside the document (https://github.com/FreeCAD/FreeCAD/issues/16187)

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Anything else?

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-09-01:
It will be necessary to take care that unsaved section views don’t mark the document as modified. Measurement has a similar issue: #13735.

pierreporte, 2024-09-01:
Do you also want that it works like Catia where sections are by default limited to a rectangle? Though unlike Catia it would cut the whole model by default.

maxwxyz, 2024-09-01:
Nt sure how section cuts work in OCC but I guess it needs some limit? It should definitely cut the entire document by default.

leoheck, 2025-09-09:
Having a result like the picture above would be a hell of an improvement.
Right now, if you have a complex assembly or even a complex object. it is quite hard to get the details since the solid looks empty. Having something like that and also allowing for measuring features while the objects are cut would be awesome.