FreeCAD/FreeCAD #16192: save method on document object not working from python script

iamthebull, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the FreeCAD GUI open any part file and modify it in any way that would mark the file as modified. A `*` appears beside the file name in the tab. Now in the python console run the command:
`App.activeDocument().save()' or `App.getDocument('_filename_').save()`
The document does not save. The `*` remains and when you close the file you will be prompted to save.

### Full version info

```shell
[code]
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * A2plus 0.4.66
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * SelectorToolbar
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bdieterm, 2024-09-01:
Try `Gui.activeDocument().save()` or `Gui.ActiveDocument.save()` instead (with "Gui" instead of "App").

iamthebull, 2024-09-01:
> Try `Gui.activeDocument().save()` or `Gui.ActiveDocument.save()` instead (with "Gui" instead of "App").

Yes that works. I suppose the App methods are to be used in console mode (ie when not working within the GUI)?

closed 2024-09-01 by iamthebull

bdieterm, 2024-09-01:
Exactly. The App methods do not require the FreeCAD GUI to be loaded.