FreeCAD/FreeCAD #16193: Save clipping plane intersection outline as object

maxwxyz, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Closely related to:
- https://github.com/FreeCAD/FreeCAD/issues/12670
 
Currently it is not possible to save the result of a clipping plane, similar to https://wiki.freecad.org/Part_Section
It should be possible to store the result outline as an object connected to the saved clipping plane (https://github.com/FreeCAD/FreeCAD/issues/16187) in the document as tree object, either the same as Part_Section result or as Sketch or Draft Wireframe object.

The result should follow what was selected to cut (https://github.com/FreeCAD/FreeCAD/issues/16190)

It should save the intersection regardless if this clipping plane is active or not.

When the section plane is updated the result should update as well.

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.4
  * Curves 0.6.42
  * dodo 1.0.1
  * fasteners 0.5.25
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
  * OpenTheme 2024.7.30
  * OpticsWorkbench 1.0.19
  * Rocket 3.3.0
  * sheetmetal 0.4.23
```

### Subproject(s) affected?

Core

### Anything else?

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-09-01:
Is this making the contours of cut parts pickable, like in Catia? I don’t mean the annoying window that display them, but them being in the 3D view as well.

maxwxyz, 2024-09-01:
@pierreporte would be both, but not in a separate window but at the clipping plane position in the 3D view.
I think this should also be optional in the clipping plane dialog.