FreeCAD/FreeCAD #16197: CAM: Creating a job with just a face as base geometry fails

opensourcecnc, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When you try to create a job with just a face as base geometry, the job creation fails with an error, because the extensions are set to 0 and the stock has no volume. You have to resort to a workaround, adding another body, setting the z extension to 1 mm then delete the initial body. 

shape = Part.makeBox(self.length, self.width, self.height, self.origin)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: height of box too small

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * AnimationFreeCAD
  * Assembly4 0.50.15
  * Curves 0.6.45
  * ExplodedAssembly-master
  * Movie 2024.2.15
  * pcb 6.2023.1
  * Render 2024.8.4
  * sheetmetal 0.4.24
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
not sure if a bug or if it works as intended @sliptonic

JohnOCFII, 2024-09-01:
From the issue posted -- even if the behavior (not allowing job without depth) is as intended, it appears the error is only displayed in the console/report window - not in a clean UI dialog box to the user.

opensourcecnc, 2024-09-01:
> it appears the error is only displayed in the console/report window - not in a clean UI dialog box to the user.
The dialog box error for the user should be an issue itself, there's almost no such thing, at least in the cam workbench