FreeCAD/FreeCAD #16198: Sketcher: chain selection by double-click doesn’t work for slots

pierreporte, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The feature introduced by #11925 doesn’t work for slots (`Sketcher_CreateSlot` and `Sketcher_CreateArcSlot`). It works for polygons, connected splines, rectangles, even rounded rectangles with a high enough radius to become slots.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
@PaddleStroke FYI

Roy-043, 2024-09-01:
I can't confirm this.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

luzpaz, 2024-09-01:
> Version: 0.22.0dev.38467 (Git) AppImage

@pierreporte please update to latest development revision and retest

pierreporte, 2024-09-02:
I still have the problem with the latest weekly:

```
OS: Ubuntu 24.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git) AppImage
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```

I follow these steps:

1. Start FreeCAD
2. Click on Parametric Part
3. Create a sketch in Part Design
4. Create a slot
5. Double click on an edge

Exiting and opening again the sketch does not solve the issue. However creating a new sketch in the body fixes it.

I just tried a third time with a new FreeCAD session and there was no problem. Same after the 4th time. I’m always doing the same thing though.

pierreporte, 2024-09-03:
I tried again some time after the last comment and the problem was back, despite doing again exactly the same actions. There is no apparent pattern.

Roy-043, 2024-09-03:
I can reproduce with this test case: #16239.

luzpaz, 2024-09-03:
Merging into #16239

closed 2024-09-03 by luzpaz

maxwxyz, 2024-09-30:
@luzpaz is this correctly closed? The liked issue seems not related to me.
I have the same issue as this one here in the RC
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.10
  * Curves 0.6.47
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.24
  * Rocket 3.3.0
  * sheetmetal 0.4.26
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```