FreeCAD/FreeCAD #16199: Exportar .stl com erro de posição em Z

Claudiobrau, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Modelo com mais de um corpo quando ajustado a posição em Z para encaixar montando uma unica peça, ao exportar esta dobrando o valor do Z na posição ficando distante o objeto sempre o dobro da distancia.

### Full version info

```shell
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
[/code]
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-01:
Please edit the issue and submit it in English. Please also try the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds if the issue is already fixed.

Roy-043, 2024-09-01:
Confirmed for V0.21, but already fixed in V0.22.

closed 2024-09-01 by Roy-043