FreeCAD/FreeCAD #16200: CAM: Total job time doesn't take into account arrays

opensourcecnc, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The total cycle time of a job doesn't take into account the array, it only calculates the time for the base operations. 
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * AnimationFreeCAD
  * Assembly4 0.50.15
  * Curves 0.6.45
  * ExplodedAssembly-master
  * Movie 2024.2.15
  * pcb 6.2023.1
  * Render 2024.8.4
  * sheetmetal 0.4.24
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-03:
@opensourcecnc can you upload an example file ? and reconfirm that this is present on 1.0 ?

luzpaz, 2025-05-31:
@tarman3 any chance you could examine this ?

tarman3, 2025-05-31:
#21303

luzpaz, 2025-05-31:
Closing as fixed by @tarman3 in #21303