FreeCAD/FreeCAD #16201: [Draft] [Regression] 0.22dev element can be moved to more than one draft layer

semhustej, 2024-09-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In 0.22dev version there seems to be regression when moving objects between layers - dragging them on another layer in the tree view. In 0.22 when the element is on an existing layer and moved to the another layer, the result is that the element is now on both layers, which is incorrect. This is not how the behavior was in 0.21.2 and also not how wiki describes the behavior. It is now also impossible to remove the element from the layer.

I am able to reproduce the behavior every time I try.

Steps to reproduce:
1. Place element one draft layer
2. Drag the element in the tree view to second draft  layer
3. The result is that the element is on both draft layers

See the behavior in the video:

https://github.com/user-attachments/assets/039a979b-db54-4afb-a17c-5beccf9a526c

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-01:
Confirmed, last good build I have:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66
```

first bad build I have is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * freecad.gears 1.2.0
  * Curves 0.6.45
  * A2plus 0.4.66
```

So somewhere between mid May and the end of June was when the regression was introduced.

Maybe look at https://github.com/FreeCAD/FreeCAD/commit/8bcbeae68b3bdad7d558cfb39e09bab798364593 and https://github.com/FreeCAD/FreeCAD/commit/7496eb94f8000a800d07a51d47b24bb4deddb8b5 initially.

Roy-043, 2024-09-02:
@semhustej Thanks for reporting.

@Syres916 Excellent detective work. It's the 2nd commit you mention.

semhustej, 2024-10-15:
Can this issue be re-opened? I found a use case, which is not yet fixed:

Element can be placed on two layers simultaneously, when it is moved from layer **placed inside a group** to different layer. Steps to reproduce:

- Open file 
[FC-duplicate-layer-placement.zip](https://github.com/user-attachments/files/17381296/FC-duplicate-layer-placement.zip)

- drag the line from Layer02 to Layer01
- the result is that the line is inside both layers - Layer01 and Layer02

See behavior in video:

https://github.com/user-attachments/assets/89491fff-3ae4-4d1a-9217-e1778447fee1

FC version:
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38946 (Git)
Build type: Release
Branch: main
Hash: e478192aa5393de9b7df190341a043f79ec6ac49
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

```

This seems like a regression. I am not able to reproduce the behavior in FC 0.21.2.

Roy-043, 2024-10-17:
#16212 assumes that layers are always in the LayerContainer. This is how it 'should' be. But, on the other hand, we do not enforce this. And the Layer Manager does find all layers in the document.

I'll improve the fix.

PR #16212 (2024-09-02): Draft: Fix regression with layer drag-and-drop behavior
Fixes: #16201

Changing the Group property type of layers to `App::PropertyLinkListHidden` also means that they longer occur in the InList of nested objects. This side-effect was missed in #14506.
commit 77539c10 (2024-09-02): Draft: Fix regression with layer drag-and-drop behavior

Fixes: #16201

Changing the Group property type of layers to `App::PropertyLinkListHidden` also means that they longer occur in the InList of nested objects. This side-effect was missed in #14506.
PR #17304 (2024-10-17): Draft: Improve layer drag-drop fix.
The fix in #16212 only checks the LayerContainer for layers. Since we do not stop users from moving layers out of that container, we should search the whole document instead.

Fixes: #16201.
commit 6c8c2632 (2024-10-17): Draft: Improve layer drag-drop fix.

The fix in #16212 only checks the LayerContainer for layers. Since we do not stop users from moving layers out of that container, we should search the whole document instead.

Fixes: #16201.
closed 2024-10-17 by yorikvanhavre