FreeCAD/FreeCAD #16205: Macro: RootObjects no longer contains all root objects

samohd, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Beginning with this build RootObjects no longer returns all root objects.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-09-02:
What commit led to this regression ?

Syres916, 2024-09-02:
The only commit I can find relevant to this report is four months ago https://github.com/FreeCAD/FreeCAD/commit/344b125a0795d1fc39db73dab315ae6b98a57e73 where it looks like the recommended code to use is `doc.RootObjectsIgnoreLinks` rather than `doc.RootObjects`

samohd, 2024-09-02:
I just tested with RootObjectsIgnoreLinks (which I was unaware of) and that shows all the Bodys.

But weirdly this morning after a lot of work on this project, particularly with VarSet... VarSet does not appear with neither RootObjects nor RootObjectsIgnoreLinks. It did with RootObjects yesterday. Assembly appears in both.

samohd, 2024-09-02:
Also note that I've only been using FC for a couple weeks, and I've installed each new nightly build as they've been released. This issue only happened with the current build indicated.

Syres916, 2024-09-02:
It's always easier for the developers if they have a file in a fault situation to be able to test against, could you please provide it?

samohd, 2024-09-02:
I presumed it would affect all documents, so I didn't provide it. I'll upload something.

samohd, 2024-09-02:
I did a test with a new project and that works as expected (attached below). So, something else went wrong that has confused the document I've been working on. I have attached it.

Someone in help forums mentioned that they loaded a variation of this project file and they couldn't change the color of a body, though I could.

This code:

doc = FreeCAD.ActiveDocument
for obj in doc.RootObjects:
	print(obj.Name)

print()
for obj in doc.RootObjectsIgnoreLinks:
	print(obj.Name)

results in:
13:24:09  Body002
13:24:09  Assembly
13:24:09  
13:24:09  Body
13:24:09  Body001
13:24:09  Body002
13:24:09  Body004
13:24:09  Assembly

[Box1x.zip](https://github.com/user-attachments/files/16839768/Box1x.zip)

This just created test project works as expected...

[Unnamed1.zip](https://github.com/user-attachments/files/16839799/Unnamed1.zip)

closed 2025-07-31 by github-actions[bot]