FreeCAD/FreeCAD #16206: Thickness shelling failed.

weety, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When designing the attached model with FreeCAD, shelling fails. There are too few shelling parameters. Arc-type shelling is not desired and is not supported by FreeCAD.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git)
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods:

3D_Printing_Tools
A2plus 0.4.62
AirPlaneDesign 0.4.0
Assembly4 0.50.6
BIM021 2021.12.0
CurvedShapes 1.0.5
Curves 0.6.22
Defeaturing 1.2.2
ExplodedAssembly
fasteners 0.5.0
fcgear 1.0.0
freecad.gears 1.0.0
Glass (Disabled)
kicadStepUpMod 10.22.9
LCInterlocking
sheetmetal 0.4.0
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-02:
Please provide an example file and steps how to reproduce

weety, 2024-09-03:
Example file:
[image]

[3D01-04.zip](https://github.com/user-attachments/files/16842447/3D01-04.zip)

arena-, 2024-09-12:
I ran into the same problem: 

When using the Thickness tool on a lofted body, the operation fails with the error:
 
<PartDesign> FeatureThickness.cpp(150): Exception on making thick solid: BRep_API: command not done
Thickness: Failed to make thick solid

The same file and operation works in 0.21.2 

On 1.0 RC 1:
[image]

On 0.21.2:
[image]

File:  
[thickness.zip](https://github.com/user-attachments/files/16977312/thickness.zip)

To recreate, open the file above (or create a simple lofted section between a circle and a rectangle - from rectangle to circle):

1. Click on the Thickness tool
2. Click on the top Face of the Rectangle
3. Click OK

AIRCAP, 2024-10-18:
sometimes a workaround:
1. Select part
2. Part -> 3D Offset
3. Boolean Cut between 3D Offset and the original Part (now it's a hollow shape)
4. Select a face to open
5. PartDesign -> SubShapeBinder
6. Part->Extrude (have it extrude downwards through the skin)
7. Boolen Cut between the skin and the extrusion -- you now have a hole where you want it (if you want it)

this allows more fine control over the skinning and also works on shapes where thickness fails (such as nurbs surfaces)

LeafMotif, 2024-12-08:
I have the same issue. The "Thickness" operation is failing when it's placed after a "Chamfer" operation. If I remove the chamfer operation, the thickness operation works. See attached file if curious. 
[Thickness fail after Chamfer.zip](https://github.com/user-attachments/files/18053883/Thickness.fail.after.Chamfer.zip)

turtle-bazon, 2025-01-13:
Thickness failed after using PartDesign_Groove, freecad 1.0.0

turtle-bazon, 2025-01-13:
Workaround doesn't work for something non-constant width. And non constant growing width. So there are too many manual work for workaround. More simple way could be using scaling object and substacting, but on complex faces it couldn't provide constant thikness.

turtle-bazon, 2025-01-13:
[image]

This simple object with failed thickness.

turtle-bazon, 2025-01-14:
[image]

I investigated a bit. If we will produce this, just cutting out upper 5 mm of prefvious then we also cant make thickness in general, but can do in particular.

turtle-bazon, 2025-01-14:
[image]

If we select bot top and bottom face then thickness fail, if we only select bottom face thickness fail. But if we select top face then thickness works!

turtle-bazon, 2025-01-14:
[image]

That is it.

turtle-bazon, 2025-01-14:
https://developer.bazon.ru/owncloud/index.php/s/pHXiCtoIViF9tlk

There are examples of several approaches in creating objects. Only on one of them thickness works.

theosib, 2025-02-08:
Here's another example of Part Thickness failing with "BRep_API: command not done".

[overture drawer.FCStd.zip](https://github.com/user-attachments/files/18720555/overture.drawer.FCStd.zip)

stefano2734, 2026-09-12:
What is the actual state in 26.3 dev with occt 7.9.3 with libpack 3.4.0 and new 8.0.1 with libpack 3.5.5?

flaviut, 2026-11-16:
@PeonyThePeon's test case was fixed in OCCT 8.0.1 (FreeCAD 26.3)

[image]

The original `3D01-04.FCStd` is still failing: silently producing invalid geometry, and [more recently](https://github.com/Open-Cascade-SAS/OCCT/commit/53706f78a8640c4407bdcc6752df13a270b159f2) an actual error "Failed to make thick solid".