FreeCAD/FreeCAD #16207: Assembly: TNP

cbenhagen, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Assemblies break on small changes due to TNP.

In this assembly the cylinder in the middle is fixed to the hole of the first body:
[image]

Adding a chamfer to the first body breaks the assembly:
[image]

Test file:
[assembly_tnp.FCStd.zip](https://github.com/user-attachments/files/16833873/assembly_tnp.FCStd.zip)

/cc @PaddleStroke

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenDark 2024.3.13
  * Manipulator 1.5.7
  * fasteners 0.5.28

----

Also happens in latest Ondsel build:
OS: macOS 14.6.1
Word size of Ondsel: 64-bit
Version: 2024.3.0dev.38472 (Git)
Build type: Release
Branch: flavor
Hash: 60c383a71ad9a470a811ccf6537156e03f180539
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Ondsel-Lens 2024.7.5.02
  * fasteners 0.5.24
  * Curves 0.6.41
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pixelwaster, 2024-09-02:
Sort of confirming for 0.22.0dev.38555, cylinder shift inside hole rather than to edge as shown.
Having issues with Ondsel on my laptop.

 
[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * AddonManager
  * Assembly
  * BIM
  * CAM
  * Draft
  * Fem
  * Help
  * Idf
  * Import
  * Inspection
  * Material
  * Measure
  * Mesh
  * MeshPart
  * OpenSCAD
  * Part
  * PartDesign
  * Plot
  * Points
  * ReverseEngineering
  * Robot
  * Show
  * Sketcher
  * Spreadsheet
  * Start
  * Surface
  * TechDraw
  * Test
  * Tux
  * Web
```

bgbsww, 2024-09-03:
What exactly about this makes it a TNP issue and not an Assembly one?

How is the cylinder fixed to the hole?  It looks to me like Sketch001 is attached to XZ_Plane001, no?

Since it is Assembly, we can't compare to 21.2 to prove or disprove, but jumping to TNP seems a stretch like #12245 which should also not be tagged or labeled TNP until there is some evidence to show it.

Yes, changing a sketch causes something to move.  And XZ_Plane001 changing seems crazy to me.  But the why might not be TNP related.  If it is, happy to chase it.

drwho495, 2024-09-05:
> What exactly about this makes it a TNP issue and not an Assembly one?
>
> How is the cylinder fixed to the hole? It looks to me like Sketch001 is attached to XZ_Plane001, no?
>
> Since it is Assembly, we [...]

I think the issue here is that the assembly workbench uses the old edge based naming method, and not the newer hash based one.

narical, 2024-09-07:
[Запись экрана от 2024-09-07 17-19-15.webm](https://github.com/user-attachments/assets/1ac35d48-ff69-4a4a-8fcb-2ca178850889)
Here's video example

maxwxyz, 2024-09-16:
@PaddleStroke FYI, is Assembly already TNP aware?

bgbsww, 2024-09-16:
It appears to be trying to, but is using a property type nothing else has ever used? https://github.com/FreeCAD/FreeCAD/issues/16259#issuecomment-2353450114

Krypto31337, 2024-09-16:
I've been playing with the integrated assembly workbench for the last few weeks and I've seen this issue quite a bit if you make even small changes.  This is about the simplest example I can make:

[image]

On the left is an assembly of 3 blocks with two fixed joints.  On the right is the same assembly if you add a chamfer to the inside edge of Component1 (bottom red block). It seems the Comp1_Comp2_Fixed joint moves to the new chamfer edge causing the blocks to shift.

[assembly_INT.zip](https://github.com/user-attachments/files/17019874/assembly_INT.zip

OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git) AppImage

ToniTen, 2024-09-17:
I have been playing with the integrates assembly workbench too, and found similar issues. I feel my example is even simpler, as discussed [here](https://forum.freecad.org/viewtopic.php?p=781870#p781870)

[AssemplyTNPIssue.zip](https://github.com/user-attachments/files/17025533/AssemplyTNPIssue.zip)

The two assembled parts are identical and link to the same Part in the same file
[image]
[image]

As you see, the joints seem to be following the name of the edge (edge12, edge14)

Steps to reproduce

```
1. Unpack the files into any directory
2. Open assembly.FCStd
3. Go to component1.FCStd
4. Add a chamfer to any of the outside edges of the body
5. Recalculate the assembly

```
FreeCAD version information
```
OS: Ubuntu Core 22 (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38768 (Git) Snap 1138
Build type: Release
Branch: main
Hash: 758674d40be45e365f4148fb2a5ea6bd6f93bc74
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fasteners 0.5.24
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * Manipulator 1.5.7
```

felixscheibe, 2024-09-17:
I built a simple assembly with a few parts, mainly using revolute joints. As soon as I applied a fillet to a random edge, the assembly became corrupted.

Testfile with issue: [Assembly-WB_Filet Problem.zip](https://github.com/user-attachments/files/17006339/Assembly-WB_Filet.Problem.zip)

Screenrecording: https://github.com/user-attachments/assets/3327db1d-bf1f-4a15-a10c-a25caf0c202e

As an alternative, I built the same model using local coordinate systems in the bodies and used them to create the joints, without any problems when modifying the parts.

Testfile with LCS in bodies: [Assembly-WB_test_lcs.zip](https://github.com/user-attachments/files/17006343/Assembly-WB_test_lcs.zip)

**Version:**
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38642 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc1)
Hash: 60a251354e203bdb924e2190fb9f5e18b48d7362
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * Manipulator 1.5.7
  * sheetmetal 0.4.25

(as requested in #16554, I added my issue here)

maxwxyz, 2024-09-17:
@chennes would you consider this a `blocker`as TNP fixes should also work in Assembly WB?

bgbsww, 2024-09-17:
Can we please stop labeling these Assembly bugs as Toponaming?  Yes, things are jumping, but they aren't TNP issues.  For example, if you look at the assembly_tnp example at the top, when you add the chamfer, you make one more piece of geometry in the sketch - Edge15.   Edge14 which is the center hole used in assembly **does not move and does not TNP**

However, Origin001 of Body001 does move, causing the visual jump.  That is a flat out Assembly bug that has absolutely nothing to do with TNP.

Please change the title, please remove the tag, and then let's see if we can't get to looking for why Assembly is jumping Origins when it should not be doing so.

bgbsww, 2024-09-17:
Now, onto diagnosis here.  It is interesting that Origin001 that matched Origin at what appears to be 0,0,0 in GCS has moved by exactly the amount to put it on the edge of the box.  This is the case even if you extend the box first.

So what appears to be happening here is that Assembly is mis-re-calculating the placement after a geometry change.  And that mis can hit either the position, the rotation or both.  You can see this clearly be checking the .Placement values in Python after it happens.

bgbsww, 2024-09-20:
After much testing

If you add a dynamic property to any documentObject of type "PropertyXLinkSubHidden" [ the one used by Reference1 and Reference2 in Assembly ],  set it to refer to something like a edge in a sketch, and then change the sketch, the value of the PropertyXLinkSubHidden properly changes to the new edge.  This property type appears to be working correctly.

However, if you do the same thing in Assembly, the Reference1/2 do not change in response.  Since we are still pointing at the same Edge/Face/Vertex name and it has moved, you get the effect shown.

bgbsww, 2024-09-20:
Okay, I think I have found a fix here.  More testing needed.

A few background points:   Assembly with PropertyLinkSubHidden instead of PropertyXLinkSubHidden works perfectly, but of course this limits what you can do with it.   Assembly is the only thing trying to use PropertyXLinks in this capacity.  And PropertyXLinks in this capacity don't work correctly - they ignore the TNP mitigation changes send to them.

If I do have the fix for PropertyXLinks correct, we'll be ready to rock and roll ...

CalligaroV, 2024-09-21:
Hi Everyone!

I was trying #16691 and I saw that it actually helps moving forward with this issue (nice job @bgbsww, as always!)
In detail I tested the files supplied with the comments in this issue and I found that all except @ToniTen's one behaves correctly (no visual jumps) **if the fillet/chamfer is applied to already existing features/sketches**

The visual jump however still happens, at least in my tests, if a new feature is added and the Tip of the Body is therefor modified.
This makes me think that to completely solve this issue some modifications are needed in the Assembly WB to handle correctly the update of the Tips of the Bodies, at least if those Bodies are used by some Joints.
@PaddleStroke, are there any method in Assembly WB that triggers when the Tips change?

Side note: about @ToniTen's example, I think that #16691 doesn't help because the last feature of `component1` is a `PartDesign::Hole` which AFAICS is not an element-aware operation (there's not even a related `OpCode` in `Part::OpCodes`)
@bgbsww, is my analysis of this particular case correct?

bgbsww, 2024-09-21:
@CalligaroV actually, in investigating the @ToniTen example, this appeared to be failing because we are trying to propogate TNP changes across document boundaries, and XLink doesn't seem to currently do that.  But we can test your hypothesis as well.  Maybe XLink does do this, just not on PartDesign::Hole ?

bgbsww, 2024-09-21:
Confirmed, TNP is missing for FeatureHole.  That's the root of that problem.

bgbsww, 2024-09-22:
A last interesting point. Component1Part.Body.Hole.Edge12 is broken and Component1Part.Body.Edge12 works, but also so does Component1Part.Hole.Edge12.  That begs the question of if we actually formatting the reference names correctly.  addSelection in JointObject.py has some fairly suspicious code that is trying to force handle TNP names, and it may be that it should be calling some standard helpers to clean up those names.  But I'm done for today, so that will need to continue tomorrow.

closed 2024-10-03 by chennes

HoWilgh, 2024-10-16:
should the above mentioned problem from [felixscheibe](https://github.com/felixscheibe) when introducing e.g. a fillet also be solved with this FIX?
I tested the current FreeCAD_weekly-builds-38977-conda-Linux-x86_64-py311.AppImage (FreeCAD 1.1.0, Libs: 1.1.0devR38977 (Git)) also with another maybe simpler model reported on https://forum.freecad.org/viewtopic.php?p=784630#p784630 and it still does not work. Should the fix be incorporated there?

luzpaz, 2024-10-16:
cc @felixscheibe can you still reproduce post-fix ?

felixscheibe, 2024-10-17:
> cc @felixscheibe can you still reproduce post-fix ?

Applying a filet to my provided example now works as espected, no messed up assembly anymore.

OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.28
  * Manipulator 1.5.7
  * sheetmetal 0.4.25

PR #16726 (2024-10-03): Assembly: Correctly build reference from selection subname
Fix #16207.   I have switched from a python based solution to a lower level c++ one.