FreeCAD/FreeCAD #16211: Toolbar items has needless paddings for built-in non-classic themes

YohDeadfall, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I noticed a while ago that a newer versions of FreeCAD has weird paddings after each tool bar button like if there should be arrows, but these just usual buttons. The problem applies to light and dark themes, and cannot be reproduced with the classic theme or without style sheets.

[image]

### Full version info

```shell
[code]
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.3.12
  * A2plus 0.4.60n
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-02:
please test the dev version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

luzpaz, 2024-09-02:
Yea, this should be a non-issue on dev. Closing.

closed 2024-09-02 by luzpaz