FreeCAD/FreeCAD #16213: Measurement: Hiding does not hide Deltas

Murmele, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a measurement and turning on the delta after hiding the measurement, the deltas are still visible
1) Create measurement
2) Save the measurement
3) Close measurement tool
4) Enable delta for measurement
5) Hide measurement

## Expected

The direct distance and the deltas get invisible

## Observed

The direct distance is anymore visible, but the deltas are **still** visible

**Measurement Visible:**

[image]

**Measurement Hidden:**

[image]

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-09-02:
@hlorus

maxwxyz, 2024-09-24:
@chennes would you consider this a blocker? Once measurements with delta are created, they cannot be hidden. Only way to remove them, is to delete the measurement object.

chennes, 2024-09-25:
@maxwxyz can you please confirm that #16819 resolves the issue for you?

maxwxyz, 2024-09-27:
can confirm, it fixes the issue

PR #16819 (2024-09-30): MeasureGui: Also hide delta measurements
This adds a SoSwitch to the globalSeparator of ViewProviderMeasureBase and connects it to the pcModeSwitch.

Fixes #16213
closed 2024-09-30 by yorikvanhavre