FreeCAD/FreeCAD #16215: CAM: Pocket and MillFace safe and clearance height movement wrong

opensourcecnc, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the pocket and millface operations movement throughout the operation when it's not cutting is done at clearance height, then the toolbit goes to safe height. After going through both heights, the toolbit is lowered at cutting position and continues the operation.
I will check the issue in other operations and modify the report if I find it happening elsewhere

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * AnimationFreeCAD
  * Assembly4 0.50.15
  * Curves 0.6.45
  * ExplodedAssembly-master
  * Movie 2024.2.15
  * pcb 6.2023.1
  * Render 2024.8.4
  * sheetmetal 0.4.24
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2025-07-20:
Probably this problem was fixed

[image]

tarman3, 2025-09-20:
Yes

closed 2025-11-29 by sliptonic

opensourcecnc, 2025-11-29:
The problem is still there. In the screenshot, the movement towards the first hole should be done at 35 mm (clearance height, before the operation), then all the rapid movement should be done at 3 mm (safe height, during the operation). Instead, all the rapid movements are done at 35 mm.

[image]

tarman3, 2025-11-29:
Please attach file

opensourcecnc, 2025-11-29:
[lead in-out entrance offset problem.20251129-182954.zip](https://github.com/user-attachments/files/23834823/lead.in-out.entrance.offset.problem.20251129-182954.zip)
Sorry, here it is. I'm using the latest build on github: [Development Build weekly-2025.11.26](https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.11.26)

tarman3, 2025-11-29:
> all the rapid movement should be done at 3 mm (safe height, during the operation).

As I know it is never be like you describe
In all operations movements between points is always at clearance height

You can use `MinTravel` to exclude retract

tarman3, 2025-11-29:
At this moment we have only one height for travels between passes and operations, it is `ClearanceHeight`
For me looks terrible, if this height also will be one for all operations in Job 
I use 'big' `ClearanceHeight` only for first operation on model
 
All will changed than we get ability to set `retract height` for travels between passes

tarman3, 2025-11-29:
Example with two `ClearanceHeight` values
One for Input and another for Output
Only first operation on model use big height
I do not need big height for travels between operations inside model

[image]

tarman3, 2025-11-29:
> wouldn't a starting point be enough

You can not set only Z
You need to set 3 coordinates
And if something changed in model, you will need to find correct position again

opensourcecnc, 2025-11-29:
I was talking about the proposed behavior in the document: clearance between operations, retract inside operations. And, if you need to change the clearance at the end of an operation to keep the tool lower, you could have a checkbox "override end clearance" where you set another value, specifically for that operation.

tarman3, 2025-11-29:
Agree
Your description looks perfect for me
#25650

opensourcecnc, 2025-11-29:
Yes, that's exactly it. One for the start would also be good (I believe the implementation is easy, once you did it for the end)

tarman3, 2025-11-29:
[image]