FreeCAD/FreeCAD #16219: CAM: Rest machining failing most of the time

opensourcecnc, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Most of the time, the rest machining option fails to properly detect what has already been milled. I have attached a file with just one example. In most cases, it still mills tiny pieces
[Unnamed3.zip](https://github.com/user-attachments/files/16839501/Unnamed3.zip)
, even though they are clearly overlapping the previous operation.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.65
  * AnimationFreeCAD
  * Assembly4 0.50.15
  * Curves 0.6.45
  * ExplodedAssembly-master
  * Movie 2024.2.15
  * pcb 6.2023.1
  * Render 2024.8.4
  * sheetmetal 0.4.24
  * woodworking 0.21.2.33771
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

opensourcecnc, 2024-09-02:
Here's another situation, when it doesn't mill anything when rest machining is enabled for the pocket after the helix, even though it should.
[Unnamed3 -err 2.zip](https://github.com/user-attachments/files/16839555/Unnamed3.-err.2.zip)

Bob-O-Rama, 2024-12-02:
Similar looking issue with this project.   There is a cylindrical pocket which is cleared by a 3.175mm end mill, then at the bottom of the pocket is a cylindrical track, about 1.5mm wide and deep cleared by a 1.0mm diameter end mill.   When Rest Machining option is set, there is no tool path.   I've attached a screen shot with / without with the operation selected.  ( These views are with the part inverted, so you can m,ore clearly see the thin circular track ( Pocket_Shape FR Grove ). 

[image]
[image]

I had though it might be die to the small volume of the track.   However the same is seen for the ( Millface Spindle Peg ) feature which is a bit larger / deeper.   Same issue, enabling Rest Machining option produce no tool path for the operation.

[image]
[image]

   
[Iomega_Alpha_FR_Applicator_v3.1.zip](https://github.com/user-attachments/files/17980802/Iomega_Alpha_FR_Applicator_v3.1.zip)

opensourcecnc, 2024-12-02:
it only fails when using offset for the initial pattern. If you change the first pattern to zigzag offset, it will create the second pass correctly, taking rest machining into account.

Bob-O-Rama, 2024-12-02:
Thanks for the suggestion.  In testing just now, that did fix the central feature, but the thin groove feature it did not.  ( I'm not complaining about your suggested workaround, BTW,  just documenting the results.  )   In playing with the width of that feature, once it got to about the same as the radius of the tool to make the initial pocket, it did work as expected.

opensourcecnc, 2024-12-03:
That is considered normal behavior and you can solve it by using extensions

FEA-eng, 2025-11-27:
@opensourcecnc Can you confirm if the issue is still relevant and should stay open?

closed 2025-11-28 by chennes