FreeCAD/FreeCAD #16222: REQUEST: VarSet rename property/label

samohd, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please make it possible to rename properties in a VarSet. This should be doable, as all other references can be renamed without issue. Likely can use the same rename code.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-09-02:
why only VarSet? The ability to edit existing properties would be useful everywhere (in the object properties panel)

Roy-043, 2024-09-03:
Renaming properties of objects should not be allowed in general. It is a sure fire way to break things.

samohd, 2024-09-03:
I don't agree. It's a very easy process... global find and replace. It's already done by other references. It can't cause any problems.

pieterhijma, 2024-09-05:
This is indeed not VarSet-specific. FreeCAD has logic to change refererences pointing to a property A to a different property B. From the top of my head, I think it could be implemented like this to rename property A to B: Create a property B with the same value as A, find the references in all open files to B and change them to point to A. Remove property B. Drawback is that if there exists a document that is not open but refers to property A, it won't get updated and the model in the document will break.

An alternative, more robust solution is to create a unique identifier for each property and the name of the property is something user-facing, more or less the same as `Name` for document object and `Label`, but more hidden from the user. The user can then rename the property but all references remain intact. This would be a big change though.

Roy-043, 2024-09-06:
Allowing users to rename any property of any object would be a recipe for total disaster. We should not even consider this.

What do you think will happen if you change the names of the properties of a Draft_Wire?

kinghat, 2024-09-06:
i think any property of any object is out of scope but still think a property variable should be able to be renamed.

samohd, 2024-09-06:
Perhaps I was not clear enough... I just want to change a Label in my Varset....

[image]

The GUI refers to these as properties. right clicking in the Varset has the "Add properties"

It's also silly that Add property, only lets you add the name of it, you then have to click again to give it a value.

samohd, 2024-09-06:
As it is now, in order to change the name, I have to add a new one, and then go find all the references to it, then delete the old one. You can rename a constraint and it automatically changes it everywhere it's used. This is no different.

kaiwas, 2024-09-25:
> Renaming properties of objects should not be allowed in general. It is a sure fire way to break things.

What nonsense? And deleting an existing property in order to recreate it with a different name and assigning a new property everywhere where the old one was, is that a good solution in your opinion?

After all, it is now impossible to even correct a typo in the description.

I understand that it is difficult to make "auto-replace" existing properties when renaming, but the very impossibility of correcting existing properties seems like a strange limitation.

kaiwas, 2024-09-25:
> Drawback is that if there exists a document that is not open but refers to property A, it won't get updated and the model in the document will break.

So if I delete an existing property and recreate it with a different name, other files will break.

But you can make a warning that this can happen, instead of limiting the user.

Please stop considering all users fools.
I think everyone is capable of limiting themselves in their actions. No need to do it for everyone.

questuk, 2024-09-29:
Hi, yes i ran into this same problem today. once you name a label you cannot edit it.
As a example, if i name the label `Test`, I cannot rename it to `Test01`

I am using ...
OS: Ubuntu 23.04 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Woodworking-master 0.21.30486
  * A2plus 0.4.67
  * Behave-Dark-Colors 0.1.1
  * FreeCAD-themes 2024.7.24
  * ProDarkThemePreferencePack 1.0.0
  * Assembly4 0.50.15

> Perhaps I was not clear enough... I just want to change a Label in my Varset.... [...]

samohd, 2024-09-29:
[questuk](https://github.com/questuk) Yeah, unfortunately there seems to be thinking that this is a feature.... rather than a bug. But fingers crossed they see the bug light and correct it. :-)

questuk, 2024-09-30:
Hi 
Using DynamicData workbench ( must first install workbench using `Tools/addon manager ` ) then it can change the varset label name, took me a few minutes to find out how ?

When installed go here ...
[image]

questuk, 2024-10-01:
Thanks **sliptonic** for adding this to the Varset project, when it's more sorted, it will be great, as I until now I used spreadsheets and https://wiki.freecad.org/Macro_EasyAlias/pl

This no longer seems to be working with FreeCAD 1.00 rc2

Thanks

sslupsky, 2024-10-01:
It would be helpful to be able to edit the VarSet property label as well as the VarSet property type.

kriswilk, 2024-10-07:
As noted elsewhere in this thread, this issue applies not just to VarSets but any custom property added to an object. In all cases, such custom properties should be painlessly renamable.

This is, in my opinion, one of the most critical bugs with VarSets/Properties. Users expect that they should be able to rename "legnth" to "length" without having to create a new variable and change ALL the references to it manually.

mnesarco, 2024-11-03:
Hi, I have a script to rename Dynamic properties. No property deletion and all references keep working, even from external files.

```python
import FreeCAD as App

def rename(object_name, prop_name, new_prop_name):
    doc = App.activeDocument()
    obj = doc.getObject(object_name)

    # Create the new property
    if prop_name in obj.PropertiesList and new_prop_name not in obj.PropertiesList:
        type_id = obj.getTypeIdOfProperty(prop_name)
        group = obj.getGroupOfProperty(prop_name)
        tooltip = obj.getDocumentationOfProperty(prop_name)
        enum_vals = obj.getEnumerationsOfProperty(prop_name)
        obj.addProperty(type_id, new_prop_name, group, tooltip, enum_vals=enum_vals)
        setattr(obj, new_prop_name, getattr(obj, prop_name))

    # Hide the old property: (do not remove but redirect)
    obj.setPropertyStatus(prop_name, 3)

    # Bind the old to the new by ExpressionEngine
    obj.setExpression(prop_name, f"{new_prop_name}", "Renamed")
    obj.recompute()

# Usage
# Hypothetical Object XX with property named lennght
rename('XX', 'lennght', 'length')

# if property is prefixed with group:
rename('XX', 'Base_lennght', 'Base_length')

```

Pros
- Nothing gets broken
- Even python macros and external files continue working if referencing the old name
- Pretty simple solution, easy to implement directly in the core

Cons:
- Old names cannot be reused to rename other properties

TommyC81, 2024-12-06:
I'm just exploring changing to FreeCad from a commercial product. So far very impressed, but ran into this limitation in my very first play through. I consider it essential to be able to rename the labels of variables/VarSet as you progress through designs. I thought I simply didn't know how to do it and came here to realize it is currently not possible.

For whatever it is worth, I'm very much in support of this feature! Thanks for great work, the 1.0 release is really good!

psommerfeld, 2024-12-21:
I'd be hard-pressed to call this a feature request. Not being able to rename a variable in a VarSet sounds like a bug!
Just used VarSets for the first time, and was surprised my misspelled variable name is set in stone!

VorTechnix, 2025-01-16:
I'm definitely in the Bug camp on this one. What I don't understand why someone thought it was a good idea (or even an acceptable developer practice) to use the property labels as property IDs in the first place?

Property labels should be just a field of the property like the value is and under the hood the property IDs should be system assigned and unchangeable (to prevent exactly the breakage problems people are afraid of here).

TommyC81, 2025-01-16:
I'm not registered on the forum and unfortunately lack the time to engage in detail. However, I fully support solid long-term solutions and believe the UUID approach (as discussed in the forum) is the best option.

VorTechnix, 2025-01-17:
> It's clear that the initial design was naive, but dwelling on blame won't solve anything.

I feel like that was directed at me 😓. Sorry. Bad dev practice is a pet peeve of mine.

The existing property system (as discussed in this thread as well as the forum) is already overly fragile and, given the history of other FOSS projects I've interacted with that turned into a patchwork of often-conflicting hacks, incremental changes/patches are more likely to make that worse than better.

I think a complete overhaul with a compatibility layer could solve the problems at hand while giving the maintainers of extensions, macros, workbenches, etc. time to transition over to the better way of doing things.

mathiaslec, 2025-03-26:
Hi,
I had the same problem with renaming properties (labels in VarSet), so I made my own macro. I post it here (with its logo) in case someone is interested. 
Is it worth posting it as an official macro?

[image]
[VarSetUpdate.txt](https://github.com/user-attachments/files/19469453/VarSetUpdate.txt)
[image]
[image]

Dan-Crane, 2025-04-08:
@mathiaslec Hi! Nice macro! Unfortunately in version 1.1 it doesn't work, when I try to run it, I get the following error:
```txt
06:36:04  pyException: Traceback (most recent call last):
  File "C:/Other/programs/FreeCAD_macro/VarSetUpdate.FCMacro", line 26, in <module>
    from PySide2 import QtWidgets
<class 'ModuleNotFoundError'>: No module named 'PySide2'
```

Have you encountered something like this?

Dan-Crane, 2025-04-08:
@mathiaslec thanks! This is work)

the only thing I changed was: `dialog.exec_()` to `dialog.exec()`

rdgarlo, 2025-04-17:
Thank you for providing this macro. This was a big reason why I preferred spreadsheets over Variable Sets. I tested this macro with FC 1.0. I created a VarSet and used its properties as constraints in a sketch. Then I ran the macro to change the name of one of the properties. It changed the name in the VarSet and also in the expression in the sketch constraint. So far, so good.

However, in the process, it created a new group in the VarSet called "Variables" and moved my variable/property to it. I would have preferred that the macro preserve the original group membership, but this is good enough for me. You have made VarSets *much* more useful. I appreciate it. :)

furgo16, 2025-04-17:
> Is it worth posting it as an official macro?

@mathiaslec thanks so much for this. I think it's worth posting it, as it provides an extremely useful workaround for a serious usability flaw, for which there is yet no solution.

Having the code in a repo will also make it easier to manage and for others to contribute.

kaiwas, 2025-04-17:
This macro can probably be improved. Right now it only analyzes one VarSet, but if we have several of them, renaming the parameters in the second one won't work.
However, it's still better than nothing

furgo16, 2025-04-17:
Indeed, I was just thinking about this while testing it now. It would be great for the macro to:

- Show a dropdown that allows selecting multiple VarSets if there are more than one.
- Autopopulate a second dropdown with the property names of the given VarSet, so that the user does not have to manually type the name of the old property.

 As an extra feature, it would also be great to be able to edit the property's Tooltip/Documentation field.

In any case, great work!

NSUBB, 2025-04-18:
I have created an edited version of @mathiaslec macro. This version allows defining the VarSet which contains the property being edited in case the project has multiple VarSets (subsequent are named VarSet001, VarSet002, etc.). I also added a Search button that lists all the properties in that VarSet. The old property name can be copied from here. Lastly, I added a user input to manually define the new Tool Tip, which was automatically being set to reference the old property name.

[VarSetUpdate_v0.1.04.txt](https://github.com/user-attachments/files/19808114/VarSetUpdate_v0.1.04.txt)

I wasn't able to figure out a way to autopopulate the existing Tool Tip, or even which property contains the string. I tried a lot of methods, which all failed.

[image]

kaiwas, 2025-04-18:
> I have created an edited version
This is just great!

I wonder - can this algorithm be redesigned to replace properties on a selected object (not VarSet)?

NSUBB, 2025-04-19:
> UPD/ By the way, with these changes the trick also works on objects.

These changes are awesome! However, I wonder if having the selected item functionality working on other objects poses any serious risk for people just mucking about? It seems a useful, but dangerous, unintentional feature.

leoheck, 2025-04-20:
The `New name`, `New Tooltip`, and the `Destination Group` should be optional.
The `Update` button should be enabled if at least one of these fields is filled and disabled if all of them are blank.
And if a field is not filled, it could reuse the existing data for that field.

NSUBB, 2025-04-21:
[VarSetUpdate_v0.3.11.txt](https://github.com/user-attachments/files/19829940/VarSetUpdate_v0.3.11.txt)

This new version resolves all issues I was able to find so far. Need some testers to see if it breaks their test projects.

New features!

1. Updates all expression references to the VarSet property name being changed!
2. Allows recreation of SAME NAME property by deleting the old before the new gets created.
3. Since same name can be used, can effectively change only the other attributes during update process.
4. Automatically retrieves and fills in fields with the current data.
5. Displays the current value and expression, if present.
6. Allows chaning PropertyType and has popup to define new value that is compatible with new type.

[image]

kaiwas, 2025-04-21:
> test projects.

The current colors are clearly made for a dark theme.
[image]

leoheck, 2025-04-21:
Can I select the property and then launch the macro with the VarSet and the property already filled up? This would be very cool.

Because the workflow would be this:

You are reading the properties and then see one that has to be updated. Then you click on de cell of the property or the cell with the value of the property that you want to update. Then you launch the macro (perhaps using some sort of shortcut) and then the fields of varset and the property name would be already filled up with the selected varset and the existing property name.

NSUBB, 2025-04-22:
> The current colors are clearly made for a dark theme.

A user on Reddit pointed this out. I've commented out the white text and changed the yellow to a dark orange. Hopefully those will be more compatible with light theme. I guess the blue is acceptable?

- Line 47 contains the yellow color #FFD700. I've been using #DC582A lately which looks a little dark on the dark theme but probably looks fine on light theme.
- Line 61 has the white color. For now I've just commented out the line because I've been thinking, maybe it should match the color of the VarSet Label.

> Is it planned to add work with the Enumeration property? Or maybe there is some way to expand the list of properties available for renaming?

I can add enumeration. I first tried to auto-populate the type dropdown with every type property but that caused the whole macro to fail as not all of them are compatible. I do not know which all are compatible, so I only hard-coded the ones the Copilot confirmed were compatible.

- The list of types being made available is at lines 254-264 right now. There are about 70 properties in the object, most of which I think are not compatible with VarSet properties.

[image]

> However, even in its current state it looks very cool! I will definitely use it. (I have not yet tested the correctness of the algorithm)

Thank you!

> Can I select the property and then launch the macro with the VarSet and the property already filled up? This would be very cool.

This was recommended by kaiwas but I chose to go a different route because I was concerned about the fact that ANY selected object could be modified which seemed like a potentially troublesome feature to me.

I'm still trying to work out some issues. Still having some problems dealing with constraints containing Expressions that reference the renamed VarSet property (even when same name). And, when the VarSet property being renamed contains an expression that is causing some issue too. I think my latest version is starting to cause more issues than fixes.

leoheck, 2025-04-22:
> This was recommended by kaiwas but I chose to go a different route because I was concerned about the fact that ANY selected object could be modified which seemed like a potentially troublesome feature to me.

You just have to ignore the selection if it doesn't belong to a VarSet.

NSUBB, 2025-04-24:
New version of macro! 

[VarSetUpdate_v0.3.14.txt](https://github.com/user-attachments/files/19881685/VarSetUpdate_v0.3.14.txt)

[image]

This version appears to be stable in all features except for changing the property type. A significant update was made to improve how expressions referencing the named VarSet property are updated. In this version, I do not receive any notifications or report messages, and sketch constraints with formulas referencing the recreated property no longer display unusual or problematic behavior. I believe the colors are now optimized for both light and dark themes. There seems to be a compromise, and the color can be easily adjusted in the macro code, where a comment indicates how to define it.

However, changing the property type remains a challenge due to the vastly different data requirements. For instance, numbers greater than 360 are truncated when switching to an Angle, and types like Bool or Enumerated have such distinct data structures that I am unsure how to handle these transitions. I attempted to implement a popup allowing users to manually change the value to match the destination type, but I could not get it to function effectively.

Additionally, transferring the current selection in the FreeCAD GUI into the macro as the property to be updated has been particularly difficult when used alongside the existing dropdown feature. Since I rarely need to change a VarSet property, I may pause development on this aspect for now. While I understand the potential value of this feature, it does not support modifying multiple properties in a single macro execution. Because, once the macro dialog box is open, the user cannot interact with the main GUI. I am uncertain whether this issue originates from the macro itself or another factor, and I do not know how to resolve it.

furgo16, 2025-04-24:
@NSUBB great work. Would you consider putting that code in a repo and making it available through Add-On Manager? It could make it easier to manage and distribute updates, and to enable contributions.

> While I understand the potential value of this feature, it does not support modifying multiple properties in a single macro execution.

I don't think it should, either. Renaming properties should, in my opinion be a Core feature (hence this issue). This macro offers a great workaround until the feature is not available in Core. My suggestion would be to keep the scope tight, so it does the one thing it's supposed to do well, until FreeCAD itself replaces it.

NSUBB, 2025-04-24:
> [@NSUBB](https://github.com/NSUBB) great work. Would you consider putting that code in a repo and making it available through Add-On Manager? It could make it easier to manage and distribute updates, and to enable contributions.

I will try to do this, but it will be my first time as I have no experience with such things. 

Thanks for the compliment!

furgo16, 2025-04-24:
I think with the continous outages folks are using less and less the forum, unfortunately. I believe the peer review step could be continued here as per https://github.com/FreeCAD/FreeCAD-macros?tab=readme-ov-file#how-to-submit-a-macro

leoheck, 2025-04-24:
Which continuous outages you say, @furgo16 ? I don't like to use the forum, for instance, it is horrible to use, the font is super small,l and the whole experience (logging/reading/posting/commenting) is annoying. Any other place is better to use, including Reddit, which is not good at all.

NSUBB, 2025-04-25:
> Thank you all for the updates, the macro is much better now!
> 
> Actually I asked on the FreeCAD forum here (https://forum.freecad.org/viewtopic.php?t=96395) what is the best way to publish it (the official FreeCAD macro repo, or an independent repo), but no one answered yet.

@mathiaslec  You are very welcome, and thanks for the compliment! I created a repository as I didn't see one in your profile. And I set it up under GNU GPL v3.0, since I didn't know any better. I hope you don't mind!
https://github.com/NSUBB/VarSet-Update

And I created a new forum post as well!
https://forum.freecad.org/viewtopic.php?p=823863#p823863

I have also experienced a lot of outages with the FreeCAD website, wiki, and forum. When discussing on Reddit, other users reported no problem, so I wonder if there is an issue specific to the USA (other user was in EU).

> I think with the continous outages folks are using less and less the forum, unfortunately. I believe the peer review step could be continued here as per https://github.com/FreeCAD/FreeCAD-macros?tab=readme-ov-file#how-to-submit-a-macro

Do you mean here, where we are discussing it now? Or at the link discussion?

rdgarlo, 2025-04-27:
@NSUBB 

I am not sure of the protocol here, but I have been discussing this over in Reddit: https://www.reddit.com/r/FreeCAD/comments/1k1jtf1/new_macro_to_change_the_name_of_a_varset_variable/ and I wanted to follow up here.

I have done some testing with macro version v0.3.14 in FreeCAD 1.0 and in AstoCAD 1.1 / 41331 running on Windows 11. I have installed the new macro and experimented with changing property names, types, tool tips, and groups. My model (same as before) has two VarSets in two different bodies, with properties in expressions within a VarSet, in the other Varset, in sketch constraints, and in operations (Pad and Revolution). I got the same results in both of the versions of FreeCAD that I used. I like the new red and blue text colors for the label name and current expression respectively. Everything worked as expected, with these exceptions:

* As you said, changing the  property type remains a challenge. When I try to change a property type from App::PropertyAngle to App::PropertyLength, then I get error messages and my property gets deleted. This one seems like a nightmare to fix (because the list of Types is *huge!*), and it is something I would rarely do, so I can live with it.

* If I change the name of a property, leave the macro window open, and subsequently Update something else, I get the error, "Property 'CircDep' not found in VarSet 'VarSet'." (where 'CircDep' is the new property name). The work-around is simple enough: Close the macro window and re-launch it after changing a property name.

> Additionally, transferring the current selection in the FreeCAD GUI into the macro as the property to be updated has been particularly difficult

This is not something that I consider necessary. I must select the VarSet and the property that I want to edit either way. Whether I do it in the model tree or in the macro window, it is the same amount of effort. As it is, I don't have to care about what is selected in the model tree when I launch the macro.

---

Because of this macro, I intend to incorporate VarSets into some of my workflows now. Again, thank you!

pieterhijma, 2025-05-01:
Thanks to everyone creating these macros! They're pretty awesome and really help filling a gap. I just wanted to let you know I'm now working on supporting this in core.

NSUBB, 2025-05-01:
That is amazing news! Thank you so much @pieterhijma

mnesarco, 2025-05-20:
Hello, I manage this with a different approach:

1. Variables as first class objects.
2. Complete GUI for Variable management.
3. Controlled recompute (auto/paused)
4. Avoid false dependencies created by multiple properties in a varsets (prevents unnecessary recomputes)
5. Import/Expor/Report
6. Create/Rename/Redefine/Show References
7. Expression bindings
8. Search...
9. Enhanced tooltips
10. Non-Modal non-intrusive UI

[image]

Quick intro video:
https://www.youtube.com/watch?v=EtGnwIlg5mU

Repo: https://github.com/mnesarco/Vars

mnesarco, 2025-06-05:
> [@mnesarco](https://github.com/mnesarco) That UI looks really cool! I especially like the export import option! I don't have time to test it right now, but I will try to soon. I'm guessing that it's placing the actual VarSet tree object in a folder to hide it? From watching the video, why did you enter the expressions that way? I never have to use <<>> or .Value in my expressions. Is that being required due to your UI, or are you just used to entering your expressions that way? In the main VarSet you can just type the name of other properties without prepending varset.xxxx. Is that method being prevented in your UI? Also, I really like the way the main VarSet docker shows the expression relationships next to each Property which has been removed in your interface. Last, there was some kind of warning listed. Is that indicating that parts of the model will break when using this?

## 1
All the created varsets are grouped and you can make all visible/invisible setting that in your preferences.

## 2
`<<Name>>.Value` is because two reasons:
1. Lookup is based on Label instead of internal name, that way FreeCAD supports renaming out of the box.
2. Vars uses varsets but in a different way than tarditional varset, with Vars each variable is a DocumentObject on its own, it provides a high gain of performance when you have many variables used in many places because it decouples fake dependencies created by traditional varsets.

## 3
I don't know what "Varset docker" is.

## 4
Which warning? can you be more specific?

leoheck, 2025-06-05:
`<<Name>>.Value` really sucks. It should have a way to keep it short. Using it as `VarSet.Height` is already long. So a simple expression is already really extensive. Now imagine adding `.value` to every single term. 

The import feature, on the video is a bit incomplete. It is not importing the group the property is. Well, maybe this still has to be implemented.

Also, I would like to see something that helps to have vars that are read only. For instance, the one with the formula could be formated different since it is not a direct input. The formula can be changed, but most of the time, it won't, specially of the formula is right. The variables with single value can be modified. So in the end you may have a big list of variables, but you can easily spot which variable you can edit and which one should be "static" for lack of better naming.

mnesarco, 2025-06-05:
You are right, this thread is for Varset renaming. I will be happy to respond to Issues related to Vars it the Vars repo. I am sorry for hijacking the thread, it was not the intention.

zbynekwinkler, 2025-06-27:
For anyone using 1.1 weeklies - this is how you can rename VarSet property today:

```python
vs = App.ActiveDocument.getObject("VarSet")
vs.PropertiesList
vs.renameProperty("group_oldname", "group_newname")
vs.PropertiesList
```

The code is meant to be pasted into the python console in freecad and shows the list of properties before and after the rename. If you have the VarSet selected in the gui, you will get a lot of messages like

```
11:10:31  QBackingStore::endPaint() called with active painter; did you forget to destroy it or call QPainter::end() on it?
```

Just select something else and then go back to the VarSet to see the changed name in the gui.

NSUBB, 2025-06-27:
Should the issue close if the user is not able to use the GUI for this functionality? I don't think many users would expect to enter the Python console to do this. And, how would they know or even know where to find instructions for accomplishing this task? Thank you so much, Pieter and everyone involved!!!!

pieterhijma, 2025-06-27:
> [@maxwxyz](https://github.com/maxwxyz) I think this issue should be closed by
> 
>     * [Gui: Add support for renaming properties #21976](https://github.com/FreeCAD/FreeCAD/pull/21976)
> 
> 
> instead of the one you linked.

Yes, I agree, it should be closed by that PR.

pieterhijma, 2025-06-27:
The Python API can be used by users as @zbynekwinkler found out, but it is currently at your own risk because the GUI cannot handle it well. That should be solved by the other PRs.

The Python function is meant for workbench developers so they have a good way to rename properties and update all references.

leoheck, 2025-07-16:
How can I rename variables now? I could not find this issue.. Right clicking the property does not display an option to rename the variable.

questuk, 2025-07-17:
same here I tried the latest weekly from 3 days ago, still no option ?

leoheck, 2025-08-09:
@pieterhijma, do you know if there has been any progress on this issue? Thank you.

pieterhijma, 2025-08-09:
Yes, PR https://github.com/FreeCAD/FreeCAD/pull/21976 is assigned to @kadet1090. He may have additional comments. If not, I think he will merge it because two reviewers have already ok'ed the PR. This gives the ability to do property rename. There was a small case that I missed as described in https://github.com/FreeCAD/FreeCAD/issues/22984. This has been fixed in PR https://github.com/FreeCAD/FreeCAD/pull/22985. I don't think many will oppose this fix.

I'm sorry that it takes so long, but FreeCAD simply never had this functionality and implementing it is more challenging than appears to be, especially when taking expressions in other documents into account and undo/redo.

closed 2025-08-10 by kadet1090

leoheck, 2025-08-14:
> I don't know what happened but perhaps the usage is a bit unintuitive. You cannot change/right-click the group name itself, but you have to select properties of which you can rename the group. I'll show a video below.

Yeah, I was aware of that. I can see this item in the menu when creating a new VarSet and Variable right now.
However, two days ago, I couldn’t see it. What I was doing was copying a Body that was linked to a VarSet into another file and then editing it there.

I’ll have to check if replicating this still causes the menu item to be missing.

Thanks for your nice reply.

[5 more replies]