FreeCAD/FreeCAD #16223: Standard view commands result is faulty camera settings if the document is empty

Roy-043, 2024-09-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Standard view commands ([Std_ViewTop](https://wiki.freecad.org/Std_ViewTop) etc.) result in faulty camera settings if the document is empty.

Scenario:
1. Start a new document.
2. Call Std_ViewTop.
3. Paste this code in the Python console:
```python
import Draft
view = Draft.get_3d_view()
print(view.getPointOnFocalPlane(0, 0))
print(view.getPointOnFocalPlane(300, 300))
```
4. Result: X and Y are zero for both points.
5. Alternatively check the camera settings with [Std_ViewIvIssueCamPos](https://wiki.freecad.org/Std_ViewIvIssueCamPos).

This is a regression compared to V0.21 and is very problematic for the commands of the Draft Workbench which rely on clicking different points in the 3D view.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38555 (Git)
Build type: Release
Branch: main
Hash: 73cb5bafe0c32c178d9b594a6ca03124483d305f
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-09-02:
From my testing basically the _focalDistance_ when there are no objects in the scene used to _100_ and is now _77.781746_.
I'll do a Git Bisect tomorrow on my older distro to make sure but this could easily go back to November 2023 (it was definitely before 21st Jan 2024), https://github.com/FreeCAD/FreeCAD/commit/52fa338bac972bbe2673de9b2d0432f70b69f154 looks like a possibility.

Rexbas, 2024-09-02:
I have not checked yet but I was also thinking that could be a possibility, note that most of that commit will be reverted once coin 4.0.3 is released. Coin 4.0.3 contains a similar fix but inside coin and will be tagged tomorrow.

Syres916, 2024-09-03:
Well (after too many hours) the Git Bisect proved I guessed the correct PR, the exact commit ref for the regression is https://github.com/FreeCAD/FreeCAD/pull/11279/commits/e327a3fd407234791b8d14b0b0e8fea934da79b5

Surely there's going to have to be FreeCAD code to allow for those users who are still on Coin 4.0.0 to 4.0.2?

Rexbas, 2024-09-03:
> Well (after too many hours) the Git Bisect proved I guessed the correct PR

Maybe next time, if you suspect a certain commit might cause a regression, just checkout that commit and the commit before it was merged. Would probably save some time :smiley: 

> Surely there's going to have to be FreeCAD code to allow for those users who are still on Coin 4.0.0 to 4.0.2?

I will add some #if directives so e327a3fd407234791b8d14b0b0e8fea934da79b5 will still compile with older versions of coin.

Rexbas, 2024-09-03:
The orthographic clipping plane fix (e327a3fd407234791b8d14b0b0e8fea934da79b5) in FreeCAD does not check if the scene is empty, the same fix in coin does check for an empty scene so I think using coin 4.0.3 will resolve this. I will check it later this week.

@Roy-043, could you also check if there are issues when using the perspective camera?

Roy-043, 2024-09-04:
The issue does not occur with a perspective camera.

Switching to perspective (Std_PerspectiveCamera) and then back to ortho (Std_OrthographicCamera) can be used as workaround.

Rexbas, 2024-09-07:
The problem is indeed resolved in Coin 4.0.3 but I have also made a PR that fixes this when compiled with older versions.

Roy-043, 2024-09-13:
There is still a remaining issue related to this. A default view can still cause problems in the Draft Workbench.
1. Open the Draft Workbench.
2. In the menu select: Edit → Preferences... And then: Draft → Grid and snapping.
3. Uncheck "Always show the grid" and uncheck "Show the grid during commands".
4. Restart FreeCAD.
5. On the Start page press the "2D Draft" button.
6. Create a [Draft Line](https://wiki.freecad.org/Draft_Line).
7. Result: the created line disappears after clicking in the 3D view.
8. Using [Std ViewFitAll](https://wiki.freecad.org/Std_ViewFitAll) solves this issue.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=90563

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38684 (Git)
Build type: Release
Branch: main
Hash: 92e31fa7ea974fe48c372f6749916d68f42dcf95
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
Installed mods:
```

Rexbas, 2024-09-15:
The line also reappears when starting to rotate the camera. Rotating the camera triggers the orthographic auto clipping fix to execute in FreeCAD code but it does not run when a new object is added to the scene. The clipping fix in FreeCAD was meant to temporary and is not perfect but just enough to reduce the worst effects. I fixed it properly in Coin, using Coin 4.0.3 resolves this.

@adrianinsaval, do you think it is possible to use Coin 4.0.3 in the weekly builds and 1.0RCs soon?

adrianinsaval, 2024-09-15:
Yes weeklies should now be using 4.0.3, I'll update the RC soon too

Roy-043, 2024-09-16:
@Rexbas Thanks for explaining. I'll close this then.

PR #16345 (2024-09-09): Gui: Fix orthographic camera standard views in an empty scene
This fixes #16223 when using Coin versions before 4.0.3.
closed 2024-09-16 by Roy-043